phrack/ShootOFF #21: Should be able to arrange targets with mouse

cbdmaul, 2015-03-26:
In the previous version, I have to select targets with the mouse and then very tediously move them with the keyboard arrows.  I'm a very lazy guy.

phrack, 2015-03-26:
I made this a bit more aggressive. I too would like to be able to get into the size and position ballpark with the mouse, then use the keyboard to get everything perfect after that if it's not already. The JavaFX canvas appears to be expressive enough that we're not talking a feature that would take me a typical ShootOFF version, unlike with the Tkinter canvas.

phrack, 2015-05-19:
Getting all of the math write given all of the different shapes in different places in a target and the various coordinate systems in JavaFX was a bitch. I'd be happy to never write a feature like this again :p.

commit 2d1e49a4 (2015-05-19): Targets can now be moved and resized with the mouse (closes #21)
closed 2015-05-19 by phrack