phrack/ShootOFF #22: Implement duck hunt animated target

cbdmaul, 2015-04-03:
A friend has demanded this feature.

phrack, 2015-04-03:
Ha! I am not going to assign it to a milestone for now, but I will label it and keep it open.

phrack, 2016-01-08:
We've mentioned this several other places, but as we link people to this issue, we should note here that we can't copy Duck Hunt. Duck Hunt is the intellectual property of Nintendo, and they do sue people who violate their IP rights (fair enough). We can be inspired by Duck Hunt, but we can't copy it.

genesisfactor, 2016-01-31:
Call it "boar hunt" instead?  I actually second this feature and would appreciate it.  my friends and i were talking about it today :)

phrack, 2016-02-07:
I appreciate the name, but it's also irrelevant without an artist making graphics. This issue will likely sit until someone with some art talent comes along and starts throwing us some graphics.

7III, 2016-02-19:
I found a couple of sites where the graphics are free:
http://www.picgifs.com/animal-graphics/geese/animal-graphics-geese-605709-604071/
http://heathersanimations.com/flying.html

While they might look a little odd flying around, a proof of concept might just get an artist annoyed enough to do the work...

phrack, 2016-02-19:
I think you're right that this is probably the best way forward. Having said that, it isn't entirely likely that it will annoy and artist and inspire them to contribute. Reliable artists willing to donate their work to open source are very far and few between in my experience.

7III, 2016-03-18:
With the flash game support, this is a bit of a moot point unless someone wants to code a version with slower ducks. Kinda hit two birds with one stone there (sorry - couldn't resist).

phrack, 2016-03-18:
@ifly53e made an exercise to practice various shotgun games. I'd like to some official word on whether or not he is going to release it before I decide what to do with this issue.

ifly53e, 2016-03-23:
When coding the trap and skeet exercise, I had to make accommodations for the lag / motion problems in 3.6 and 3.7.  For example, I had to create an object with a left and right target and then add trailing shapes behind the gif so that a hit could be recorded when the target was being moved.  I think that when 3.8 solves the lag issue, then the code could be revisited and released in an optimized state.

cbdmaul, 2016-03-23:
That's a different issue.  What you are referring to is the fact that the projection lags behind what we're sending to the arena.  That is an unavoidable delay in processing.  It runs about 100ms, but certain projectors may lag up to 300ms due to their own video processing.  I have added code to the autocalibration that can estimate this delay roughly, but that is as far as it goes for us handling it as of right now.  Nothing is planned for 3.8 to handle that delay.

phrack, 2016-03-23:
Actually, lagging a target behind a bit seems like a pretty decent approach. It's high performance and honestly, no one will notice if it's a smidge off while a target is moving around.

ifly53e, 2016-03-23:
Is that calculation available to exercises via a public method?  I guess it would be possible to make "tails" in various sizes and then pick the best size (.target file) depending on the projector delay.

The problem comes in for me when shooters can use lead to hit the target, or hit the target on the screen exactly and then rely on the tail for the hit.  A moving target can get pretty big with this approach.  It was off enough for me to question the accuracy of the exercise because the dot on the screen is in the sky when pulling lead, not the target.

phrack, 2016-03-23:
The time delay calculation is currently disabled and not even used because it was causing auto-calibration to fail. This is what was causing that problem in 3.7 where you'd get the purple square then if you stopped and restarted calibration it would auto-calibrate instantly.

cbdmaul, 2016-03-23:
I created #416 for this

phrack, 2016-07-17:
I am closing this because the new clays buster exercise fills the role well enough IMO.

closed 2016-07-17 by phrack