phrack/ShootOFF #85: Reopening shootoff causes camera issues leading to exceptions

cbdmaul, 2015-05-31:
Exception in thread "Thread-8" com.github.sarxos.webcam.WebcamLockException: Webcam FaceTime HD Camera (Built-in) 0 has already been locked
	at com.github.sarxos.webcam.WebcamLock.lock(WebcamLock.java:286)
	at com.github.sarxos.webcam.Webcam.open(Webcam.java:265)
	at com.github.sarxos.webcam.Webcam.open(Webcam.java:232)
	at com.shootoff.camera.CameraManager$Detector.run(CameraManager.java:91)
	at java.lang.Thread.run(Unknown Source)
Exception in thread "JavaFX Application Thread" java.lang.NullPointerException
	at javafx.embed.swing.SwingFXUtils.toFXImage(Unknown Source)
	at com.shootoff.gui.ImageCell.fetchWebcamImage(ImageCell.java:99)
	at com.shootoff.gui.ImageCell.lambda$0(ImageCell.java:64)
	at com.shootoff.gui.ImageCell$$Lambda$359/1687426556.run(Unknown Source)
	at com.sun.javafx.application.PlatformImpl.lambda$null$170(Unknown Source)
	at com.sun.javafx.application.PlatformImpl$$Lambda$48/771230107.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$171(Unknown Source)
	at com.sun.javafx.application.PlatformImpl$$Lambda$47/237061348.run(Unknown Source)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(Unknown Source)
	at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at com.sun.glass.ui.win.WinApplication.lambda$null$145(Unknown Source)
	at com.sun.glass.ui.win.WinApplication$$Lambda$36/2117255219.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
Exception in thread "JavaFX Application Thread" java.lang.NullPointerException
	at javafx.embed.swing.SwingFXUtils.toFXImage(Unknown Source)
	at com.shootoff.gui.ImageCell.fetchWebcamImage(ImageCell.java:99)
	at com.shootoff.gui.ImageCell.lambda$0(ImageCell.java:64)
	at com.shootoff.gui.ImageCell$$Lambda$359/1687426556.run(Unknown Source)
	at com.sun.javafx.application.PlatformImpl.lambda$null$170(Unknown Source)
	at com.sun.javafx.application.PlatformImpl$$Lambda$48/771230107.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$171(Unknown Source)
	at com.sun.javafx.application.PlatformImpl$$Lambda$47/237061348.run(Unknown Source)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(Unknown Source)
	at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at com.sun.glass.ui.win.WinApplication.lambda$null$145(Unknown Source)
	at com.sun.glass.ui.win.WinApplication$$Lambda$36/2117255219.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)

phrack, 2015-05-31:
Yea, this is because two instances of ShootOFF are open. This needs a better error message for the user so I will call this an enhancement.

phrack, 2015-05-31:
f8148a172c507f936c250a5486c865a78585941d

closed 2015-05-31 by phrack