phrack/ShootOFF #107: Needs more improve shot detection

cbdmaul, 2015-06-05:
We need testcase support for improving shot detection:

1. Record the video stream in ShootOFF
2. Replay saved video streams in ShootOFF
3. Basic testcase support for checking the results of shot detection during video replays
4. Any other diagnostic displays that will help us

Our problem with shot detection is that bright areas of the screen are now ignored by our bloom filter.

1. We need to stop camera auto brightness detection, or account for it somehow
2. We need to average the brightness of the screen.  I think this needs to be done using the bloom filter data.  If a given pixel is consistently “bright” it should be stepped down until it no longer triggers on bright.  This is a balancing act, of course.  If it stops being bright it needs to stop being adjusted.
3. Leverage our bloom filter data for more, right now we have more data on noise than we actually use.

phrack, 2015-06-07:
OK, if you are debug mode you can now record a feed by right clicking it and hitting "Start Recording". This will output a debug message like:

[JavaFX Application Thread] DEBUG com.shootoff.camera.CameraManager - Writing Video Feed To: /home/phrack/ShootOFF/Default.mp4

To stop recording either right click the feed again and hit "Stop Recording" or just close ShootOFF.

I am not planning for this to ever be exposed outside of debug mode. If we need to record in the future we will indirectly expose it some other way (e.g. starting some kind of training session will start recording).

phrack, 2015-06-07:
PS. The codec used is h264 for now. This is the same one used by YouTube. I chose it because it gets reasonably small files with basically no quality loss (for testing, we need the image to be true).

phrack, 2015-06-07:
OK, you can use videos recorded in prior ShootOFF sessions in unit tests now and ShootOFF will behave like a user was running that exact same session again (assuming default configuration!). 

Any unit test using a video should appear in TestCameraManager. There is one example test in there of a video I took with two shots and no interference (i.e. a very easy case).

phrack, 2015-07-10:
This issue is closed because #115 and #116 are more specific items for the points that have not yet been addressed in this issue.

closed 2015-07-10 by phrack