phrack/ShootOFF #109: Enhancements to projection area handling

cbdmaul, 2015-06-07:
Two closely related enhancements

1.  Option to disregard shots outside of the calibrated area

2.  Option to crop the video display to the calibrated area (Some kind of "undo calibration" option would be necessary to readjust the projection area)

We may want to adjust the shot detection algorithm to only apply to the calibrated area, to reduce processing and memory requirements for less powerful systems.  This could be left for later optimization.

phrack, 2015-06-07:
We could crop the picture to the calibrated area right when it comes off the camera. This would give us the optimization for free basically.

phrack, 2015-06-08:
Should probably be radio buttons:

- Only detect in calibrated region
- Crop to calibrated region

The second option is a superset of the first because the easiest way to do this is to just crop the image right off the camera.

closed 2015-07-05 by phrack

phrack, 2015-07-05:
When you start calibrating now two radio buttons are placed on the feed you are calibrating on:

- "Only detect in projector bounds"
- "Crop feed to projector bounds"

By default neither is selected, meaning everything will function as it did before this feature existed. If you select the first one, you see the entire feed still, but the copy of the feed that is processed for shot searching is cropped right before this processing takes place, thus shots can only be detected in the projection bounds because the rest literally doesn't exist. If you select the second one, the feed is cropped to the projection bounds as soon as calibration stops while images come off the camera, and by extension shots can only be detected within said bounds for the same reasons as before. These transformations can be undone to get the default behavior again by simply recalibrating and not selecting an option.

commit 0290b613 (2015-07-05): Can now choose to only detect shots in calibrated regions or fully crop the feed to the calibrated region as soon as images are pulled from the camera. This closes #109