phrack/ShootOFF #115: Improve camera detection hardware settings

phrack, 2015-07-09:
We did a number of tests with a Microsoft HD-3000 and a PS3Eye and we found conclusively that things work best with the brightness all the way down and auto white balance off. If we can turn off auto brightness we should, but it seems to not be well supported. Changing contrast did nothing.

cbdmaul, 2015-07-12:
For making adjustments in software:

I can't easily diagnose what's wrong with the software image adjustments because I have no feedback except for the stream debugger on live streams.  I need to see what the algorithm is actually doing to the frames, from videos, to see what the problem is.  I need a live testing view of the image enhancements.

Right now, I can easily make a video look like shootoff should have no problems with it by lowering the brightness very low and upping the contrast/saturation a bit, in vlc.  But I get odd results when I run it in a testcase in shootoff and I have no feedback to figure out why.

As for your lum/red changes.  I think lowering lum is good.  I think lowering red when too high AND upping red when it is too low may be the way to go.  But I can't really say for sure until I understand why I can't get it to work with certain videos that look okay

phrack, 2015-07-12:
I suggest adding strategic print statements and saving strategic frames to a file then removing them as you collect the data you need. I've been keeping this kind of data and some notes outside of the repo myself.

You can save a frame to a file like:

ImageIO.write(currentFrame, "png", new File("frame_1.png"));

You'll get a squiggly line for ImageIO because it isn't imported. Hover over it to get the quick fix to import it.

What you will find when you do this is that even a seemingly innocent print statement can slow things down substantially (e.g. that test running in 2 seconds before is now 30 seconds), so you will want to remove them as soon as you get the data you need. We haven't exposed everything we can effeciently expose, but there are also important calculation outputs in the guts of the adjustments where the code to expose it is an order of magnitude slow down. 

On thing that makes me nervous about changing color temperatures is that if we raise red too much we risk breaking green shot detection.

phrack, 2015-07-16:
The code in webcam_pixeltransformer ultimately closes this issues, but there are still three things to do:

1. Preserve the webcam frame that comes off the camera to ensure true color detection (right now some greens turn red to the colors being heated up).
2. Remove unnecessary controls from the stream debugger (bloom filter controls are no irrelevant).
3. Currently pixels are being dimmed. Try turning them straight black instead as there is likely no reason to just dim them.
4. Reduce the number of pixel loops. Currently there are 3 times every pixel in a frame is looped through for no reason and this is really slowing down processing.

phrack, 2015-07-18:
Primary outstanding issue that the code that detects the center sometimes fails to find shot width or height and that causes some good shots to be rejected.

commit eb26d9ba (2015-07-19): Improved shot center approximation to close #115
closed 2015-07-19 by phrack