phrack/ShootOFF #116: Shot detection delay in CameraManager is messing up shot detection

phrack, 2015-07-09:
It's slowing down the camera to the point that it is missing shots it should see. Remove the delay, but calculate the FPS and if we can't keep up, warn the user.

cbdmaul, 2015-07-10:
To expand a bit, this is one of those bugs where not enough debugging information leads us astray while testing for other things.  Rather than our shot detection code being a cause of bad shot detection, it was a symptom of bad frame rate.

We need to give ourselves the tools to recognize bad framerate and we need to give the user feedback when the framerate is too low for shootoff to work correctly.

phrack, 2015-07-10:
Amen. The easy part of fixing this bug is removing the unnecessary delay, the "fun" part will be accurately calculating the real frame rate we are getting. The bloom filter initialization code would be the best place to determine the "ideal" figure given the reduced processing done at that time.

phrack, 2015-07-10:
The commit I just pushed did four things:

1. It removed the buggy and unnecessary webcam delay
2. It made it so that ShootOFF will only attempt to look for a shot if it actually got a new frame from the camera. Before it could potentially search the same frame several times.
3. It now times bloom filter initialization and counts the number of webcam frames seen during that initialization process. This data is then used to calculate the fps. 
4. It fixed a super minor bug in the config class where methods weren't called in the right order. 

For 3, in debug mode you should get an output now like:

[Thread-7] DEBUG com.shootoff.camera.CameraManager - Finished initializing bloom filter (10 frames in filter): Enabling Shot Detection
[Thread-7] DEBUG com.shootoff.camera.CameraManager - Bloom filter initialized in 1450 ms with 8 FPS from the camera

8 FPS for my camera is correct based on my tests with other applications. The camera says it can do up to 30 FPS, but that's not what it does by default on Linux.

FPS is calculated as follows

bloom_filter_init_time = current_miliseconds - bloom_filter_init_milliseconds
fps_multiplier = 1000 / bloom_filter_init_time
fps = frames_seen * fps_multiplier

Can you try this with you PS3Eye and see if the output is reasonable? You may need to open a different app that measures the FPS to see what the default actually is. Again, it turns out that the advertised FPS is not necessarily the driver default. On Linux, guvcview will tell you the default FPS in the title of the window showing the webcam feed.

phrack, 2015-07-11:
I crunched some numbers and 5 FPS is about bottom of the barrel for decent but not perfect shot detection. If the FPS drops below that while shots are being detected you'll get a log message and a popup telling you that some shots may be missed and that you should try closing some other programs that are running.

I found that webcam capture actually calculates the webcams FPS in real time after reading the source code so I made the code use that now instead of trying to calculate it myself.

commit 39cb67a8 (2015-07-11): Switched to using FPS calculated real time by webcam capture. Now shows a warning if FPS is too low for reliable shot detection. This closes #116
closed 2015-07-11 by phrack