phrack/ShootOFF #131: Layout issues on high res displays

cbdmaul, 2015-07-21:
The main window still has problems with layout on high res / dpi screens

a 640x480 webcam display is tiny on a 4k display, which causes issues with the things we put directly into the cam feed.

[image]

phrack, 2015-07-21:
Hopefully someone rich enough to own such a fancy pants display will fix this and confirm the fix :p.

cbdmaul, 2015-07-30:
I'm actually not sure what the problem is.  This doesn't happen on my mbp retina, only on the dell, which I can't test anything on.

phrack, 2015-07-30:
Windows 8 on both? The Mac one is probably wrong when anyone nicknamed cbdmaul is not looking just to screw with you.

cbdmaul, 2015-07-31:
Both are high res on windows 8.1.  I'll need to compare DPI and resolution to see if that's the problem.  the mpb retina is 2880x1800 at default dpi and seems to look okay.

phrack, 2015-07-31:
You could try to relative positioning like you did with the preferences window. It might be more of a PITA with this window because I've found the list that shows the shots to be finicky about its size.

cbdmaul, 2015-08-03:
I'm still not sure what the problem is here.  In that screenshot, the webcam feed area is way smaller than it should be.  My dell is 3200x1800, windows 8.1, with the text size increased.  The 640x480 area is still large enough that the text fitting isn't a problem.

I don't know what conditions caused the problem shown in the screenshot.  It's definitely still a possible issue though.

phrack, 2015-08-03:
It kind of looks like the feed is the right size, but the shot list is just partially covering it while also stretching the window horizontally. Maybe the shot list is the problem to focus on? Maybe just it needs relative sizing and positioning?

cbdmaul, 2015-08-03:
That can't be right because the controls on the feed go off the edge.

phrack, 2015-08-03:
To me it looks like they are actually being covered by the list, not that they are going off the edge of the feed. Just like the bottom left corner of the list is covered by the reset button.

phrack, 2015-08-03:
PS. A quick test would be to just hide the list by editing the XML. Don't remove it or things will break, just set its visibility to false. Does the feed look right then?

cbdmaul, 2015-08-03:
Even on my mbp, where it looks right, the calibration controls go off the edge of the feed.  so that's a problem regardless of what's causing the other issues here.
[image]

phrack, 2015-08-03:
Weird. So yea, I guess it would be nice to know why the shot list blows up in size and why the fonts get so much bigger. The shot list is in a scroll pane that has a preferred width set but no height at all, which just makes it that much more confusing that both dimensions are changing.

cbdmaul, 2015-08-03:
The fonts get bigger because that's what happens on a high DPI screen.  We're not the only app that has problems with that.

1. The columns in the shot list should be set to auto column width.  That will make them as wide as the header text, which will fix that issue.

2. The reset button floats over the webcam feed and shot list if the "Reset" text is too big.  Need a bigger gap between them.  I don't know how to do that.

3. I suggest a smarter way of presenting the radio buttons to the user.  Maybe just make those options check marks under the "Projector" menu? That will eliminate this text size problem.

phrack, 2015-08-03:
Honestly, I don't really know how to do any of these things either. You're attempt is just as good as mine. All I'd do is open the fxml file in SceneBuilder 2.0 and bumble around until it seemed to work: 

http://www.oracle.com/technetwork/java/javase/downloads/javafxscenebuilder-1x-archive-2199384.html

Well, except for the radiobuttons, which are added programmatically, but that's the least bad part of this bug IMO.

cbdmaul, 2015-08-03:
Alright I'll see if I can get the scenebuilder to work for me at some point.

This is what we need for the auto header adjustment:

table.setAutoResizeMode(JTable.AUTO_RESIZE_ALL_COLUMNS);

I just had no clue how to add that to your code.

phrack, 2015-08-03:
That looks like the Swing version of the table. We are using the TableView in JavaFX:

https://docs.oracle.com/javase/8/javafx/api/javafx/scene/control/TableView.html

You'd do something like that by going into ShootOFFController.java and adding a new line below line 210 like:

shotTimerTable.WHATEVER_HERE

cbdmaul, 2015-08-04:
1. Fixing reset button:

Remove 'prefHeight="35.0"' from GridPane that contains reset button (Line 79, ShootOFF.fxml)

2. Fixing columns:

This one is tough.  I haven't found a universal solution yet.  But this works well enough on all resolutions/DPIs.  Replace "setPrefWidth" calls in shootoffcontroller.java with these calls:

timeCol.setMinWidth(85);
laserCol.setMinWidth(85);

3. Radio buttons

This needs to be redesigned to be fixed.

4. BONUS FIX: TargetEditor window cuts off the toolbar.

Remove 'prefWidth="622.0"' from ToolBar line in TargetEditor.fxml (Line 11)

5. BONUS FIX #2: SessionEditor has wacky controls.

Remove "pref" options and add 'HBox.hgrow="ALWAYS"' to both ListViews (lines 10 and 24)

Add 'minHeight="-Infinity" minWidth="-Infinity"' to both Buttons (Lines 16 and 17)

phrack, 2015-08-04:
I'll plan to try these tonight. Do you consider this issue closed if I do these or do you want me to just wait to do anything until you find a universal solution for the columns?

cbdmaul, 2015-08-04:
I think that it's a suitable solution for the time being.  Even scenebuilder had display problems at the resolution and text size that I was testing it with.  I don't think anyone is going to be jacking up their font sizes that much, the text was so huge I was blinding myself working on it (It gave me a headache).

The only remaining issue is the radio buttons in the camera feed.  They are cut off even at a normal font size on a 4k display, so it is more of a usability issue than the others were.

phrack, 2015-08-04:
OK, I'll go ahead and close this issue tonight most likely. I'll probably just stack the radio buttons vertically instead of putting them side by side as a quick fix. They could go under the Projector menu to let people tweak this particular setting at any time, but I am also concerned about that menu getting too busy as we add even more stuff later. If it were something I believe people want to tweak a lot, I would, but if I had to bet, I'd bet that most people will just take the default every time.

Minor rant, but SceneBuilder having this issue too tells me that JavaFX should really handle DPI issues internally unless the dev is doing something off the beaten path. There is no reason a dev that is just dragging and dropping controls should have to know anything about DPIs or various monitor resolutions to make the app look right in different environments. All Java GUI toolkits have the same problem, they make hard things possible but at the expense of making easy things hard and punting issues to the dev that the dev should only have to worry about if they are doing something really custom.

commit ea46834a (2015-08-04): Fixed various layout issues on high DPI displays. Closes #131
closed 2015-08-04 by phrack