phrack/ShootOFF #134: Add backgrounds for the projector arena.

phrack, 2015-07-26:
Would be nice to start with 4-5 scenes:

- Desert
- Forrest
- Mountains
- Savannah
- Something else...

phrack, 2015-07-27:
Possible default backgrounds:

https://commons.wikimedia.org/wiki/File:Trandumskogen_Tank_shooting_range.jpg
https://commons.wikimedia.org/wiki/File:Shooting_range_near_Pittsburgh_-_outside_12.JPG
https://en.wikipedia.org/wiki/File:DragoonMountains.JPG

cbdmaul, 2015-08-02:
I tried these out today.

I liked the shooting range the best.  It felt the most natural.  BUT the flag is really distracting due to it being in the foreground.  If there is a similar picture without the flag, or photoshop out the flag, that would be better.

The tank range isn't bad either.  Just doesn't feel like a shooting range.

The mountains didn't feel right, too much depth.  Shot color was a problem on the sky as well.  red laser + blue sky = green shot detection

I think ones with the depth of a shooting range will have the best feel.  This leads me to believe that savannas, forests, etc. will work the best (other than actual shooting ranges of course).

phrack, 2015-08-02:
Please feel free to suggest pictures that have appropriate licenses. You can tell Google images to only show images that allow resuse under Search Tools.

cbdmaul, 2015-08-02:
I took a quick look at wikimedia commons

I like this one.  We can crop it down to remove the sky.  Might want to change white target to brown as well.
https://commons.wikimedia.org/wiki/File:Gun_Range_100_Yards.JPG

This is a decent indoor one.
https://commons.wikimedia.org/wiki/File:25m_Raumschiessanlage.jpg

This is a good african savanna one, as the sky has muted colors already
https://commons.wikimedia.org/wiki/File:Kiang_West_savanna.jpg

I think this one would work for a forest shot if we dim the brightness or mute the brightness at the sky regions.
https://commons.wikimedia.org/wiki/File:Epping_Forest_Centenary_Walk_2_-_Sept_2008.jpg

phrack, 2015-08-02:
OK, cool, thx! I will pick my favorites when I go to close this issue.

phrack, 2015-08-02:
I am going to go ahead and include the savanna, indoor range, and gun range 100 yards as defaults. I don't feel like the tank range has enough space for targets and the others require more editing than I care to do at the moment.

commit 82e26288 (2015-08-02): Backgrounds can now be set for the projector arena to close #134
closed 2015-08-02 by phrack