phrack/ShootOFF #139: Exception in thread "JavaFX Application Thread" java.lang.ArrayIndexOutOfBoundsException: Coordinate out of bounds!

cbdmaul, 2015-07-30:
Shooting around using click to shoot, testing the resizing targets, I get this exception once in a while.  I am clicking in the image area, doesn't seem to be much logic or consistency to it.  just happens every 20 or 30 shots.

Exception in thread "JavaFX Application Thread" java.lang.ArrayIndexOutOfBoundsException: Coordinate out of bounds!
	at sun.awt.image.IntegerInterleavedRaster.getDataElements(Unknown Source)
	at java.awt.image.BufferedImage.getRGB(Unknown Source)
	at com.shootoff.gui.CanvasManager.checkHit(CanvasManager.java:289)
	at com.shootoff.gui.CanvasManager.addShot(CanvasManager.java:209)
	at com.shootoff.gui.CanvasManager.lambda$1(CanvasManager.java:107)
	at com.shootoff.gui.CanvasManager$$Lambda$106/1719087755.handle(Unknown Source)
	at com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(Unknown Source)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(Unknown Source)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(Unknown Source)
	at com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(Unknown Source)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(Unknown Source)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(Unknown Source)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(Unknown Source)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(Unknown Source)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(Unknown Source)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(Unknown Source)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(Unknown Source)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(Unknown Source)
	at com.sun.javafx.event.EventUtil.fireEventImpl(Unknown Source)
	at com.sun.javafx.event.EventUtil.fireEvent(Unknown Source)
	at javafx.event.Event.fireEvent(Unknown Source)
	at javafx.scene.Scene$ClickGenerator.postProcess(Unknown Source)
	at javafx.scene.Scene$ClickGenerator.access$8100(Unknown Source)
	at javafx.scene.Scene$MouseHandler.process(Unknown Source)
	at javafx.scene.Scene$MouseHandler.access$1500(Unknown Source)
	at javafx.scene.Scene.impl_processMouseEvent(Unknown Source)
	at javafx.scene.Scene$ScenePeerListener.mouseEvent(Unknown Source)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(Unknown Source)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.lambda$handleMouseEvent$350(Unknown Source)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$$Lambda$292/1728076957.get(Unknown Source)
	at com.sun.javafx.tk.quantum.QuantumToolkit.runWithoutRenderLock(Unknown Source)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.handleMouseEvent(Unknown Source)
	at com.sun.glass.ui.View.handleMouseEvent(Unknown Source)
	at com.sun.glass.ui.View.notifyMouse(Unknown Source)
	at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at com.sun.glass.ui.win.WinApplication.lambda$null$145(Unknown Source)
	at com.sun.glass.ui.win.WinApplication$$Lambda$36/2117255219.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)

phrack, 2015-07-31:
It's stupid that this exception doesn't print out what the attempted coords were I'll see if I can dupe.

phrack, 2015-07-31:
Closed 01ff829a64b85bb58be335b7e8f85e832580c1e8

closed 2015-07-31 by phrack