phrack/ShootOFF #150: IndexOutOfBoundsException in session viewer

cbdmaul, 2015-08-02:
Opening session viewer with three sessions, clicking third session, clicking on third shot, causes this exception for me.

Exception in thread "JavaFX Application Thread" java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
	at java.util.ArrayList.rangeCheck(Unknown Source)
	at java.util.ArrayList.get(Unknown Source)
	at com.shootoff.gui.SessionCanvasManager.doEvent(SessionCanvasManager.java:61)
	at com.shootoff.gui.controller.SessionViewerController.updateEvents(SessionViewerController.java:194)
	at com.shootoff.gui.controller.SessionViewerController.access$12(SessionViewerController.java:179)
	at com.shootoff.gui.controller.SessionViewerController$3.changed(SessionViewerController.java:108)
	at com.shootoff.gui.controller.SessionViewerController$3.changed(SessionViewerController.java:1)
	at com.sun.javafx.binding.ExpressionHelper$SingleChange.fireValueChangedEvent(Unknown Source)
	at com.sun.javafx.binding.ExpressionHelper.fireValueChangedEvent(Unknown Source)
	at javafx.beans.property.ReadOnlyObjectWrapper$ReadOnlyPropertyImpl.fireValueChangedEvent(Unknown Source)
	at javafx.beans.property.ReadOnlyObjectWrapper.fireValueChangedEvent(Unknown Source)
	at javafx.beans.property.ObjectPropertyBase.markInvalid(Unknown Source)
	at javafx.beans.property.ObjectPropertyBase.set(Unknown Source)
	at javafx.scene.control.SelectionModel.setSelectedItem(Unknown Source)
	at javafx.scene.control.MultipleSelectionModelBase.lambda$new$34(Unknown Source)
	at javafx.scene.control.MultipleSelectionModelBase$$Lambda$78/643266144.invalidated(Unknown Source)
	at com.sun.javafx.binding.ExpressionHelper$SingleInvalidation.fireValueChangedEvent(Unknown Source)
	at com.sun.javafx.binding.ExpressionHelper.fireValueChangedEvent(Unknown Source)
	at javafx.beans.property.ReadOnlyIntegerWrapper$ReadOnlyPropertyImpl.fireValueChangedEvent(Unknown Source)
	at javafx.beans.property.ReadOnlyIntegerWrapper.fireValueChangedEvent(Unknown Source)
	at javafx.beans.property.IntegerPropertyBase.markInvalid(Unknown Source)
	at javafx.beans.property.IntegerPropertyBase.set(Unknown Source)
	at javafx.scene.control.SelectionModel.setSelectedIndex(Unknown Source)
	at javafx.scene.control.MultipleSelectionModelBase.select(Unknown Source)
	at javafx.scene.control.MultipleSelectionModelBase.clearAndSelect(Unknown Source)
	at javafx.scene.control.ListView$ListViewBitSetSelectionModel.clearAndSelect(Unknown Source)
	at com.sun.javafx.scene.control.behavior.CellBehaviorBase.simpleSelect(Unknown Source)
	at com.sun.javafx.scene.control.behavior.CellBehaviorBase.doSelect(Unknown Source)
	at com.sun.javafx.scene.control.behavior.CellBehaviorBase.mousePressed(Unknown Source)
	at com.sun.javafx.scene.control.skin.BehaviorSkinBase$1.handle(Unknown Source)
	at com.sun.javafx.scene.control.skin.BehaviorSkinBase$1.handle(Unknown Source)
	at com.sun.javafx.event.CompositeEventHandler$NormalEventHandlerRecord.handleBubblingEvent(Unknown Source)
	at com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(Unknown Source)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(Unknown Source)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(Unknown Source)
	at com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(Unknown Source)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(Unknown Source)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(Unknown Source)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(Unknown Source)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(Unknown Source)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(Unknown Source)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(Unknown Source)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(Unknown Source)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(Unknown Source)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(Unknown Source)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(Unknown Source)
	at com.sun.javafx.event.EventUtil.fireEventImpl(Unknown Source)
	at com.sun.javafx.event.EventUtil.fireEvent(Unknown Source)
	at javafx.event.Event.fireEvent(Unknown Source)
	at javafx.scene.Scene$MouseHandler.process(Unknown Source)
	at javafx.scene.Scene$MouseHandler.access$1500(Unknown Source)
	at javafx.scene.Scene.impl_processMouseEvent(Unknown Source)
	at javafx.scene.Scene$ScenePeerListener.mouseEvent(Unknown Source)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(Unknown Source)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.lambda$handleMouseEvent$350(Unknown Source)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$$Lambda$279/1202539729.get(Unknown Source)
	at com.sun.javafx.tk.quantum.QuantumToolkit.runWithoutRenderLock(Unknown Source)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.handleMouseEvent(Unknown Source)
	at com.sun.glass.ui.View.handleMouseEvent(Unknown Source)
	at com.sun.glass.ui.View.notifyMouse(Unknown Source)
	at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at com.sun.glass.ui.win.WinApplication.lambda$null$145(Unknown Source)
	at com.sun.glass.ui.win.WinApplication$$Lambda$36/2117255219.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)

phrack, 2015-08-02:
Please upload sessions.

cbdmaul, 2015-08-02:
https://dl.dropboxusercontent.com/u/56848390/2015-08-02%2018.05.06.xml
https://dl.dropboxusercontent.com/u/56848390/2015-08-02%2018.03.48.xml
https://dl.dropboxusercontent.com/u/56848390/2015-08-02%2018.02.39.xml

phrack, 2015-08-03:
OK, so this is an interesting wrinkle and the kind of thing that has made me conservative about complicating things too early.

What's going on here and in #149 is that you added a target, then you started recording the session, then you performed an operation on the target. The problem is that a target only has an "identity" when it is added. After that, it is referred to by its index in the target list not by a more specific identity. It's easy to track this index and keep it from getting corrupted because things are not happening concurrently, but it has the problem that without the target added event you're SOL. A target wasn't showing before even though you had the target moved event because there was no way for the viewer to even know which target it was without the target added event. In this particular case it's trying to look up the target by the index, but of course that index doesn't exist because there was no target added event to create one.

There are two potential ways to fix this off the top of my head:

1. Don't record events for targets added before the user started recording. This is easy to do, but it would probably confuse some people (e.g. why isn't this target showing up in my viewer!?).
2. Maintain more metadata along with the targets so that in cases like this the viewer can say, "Oh, it looks like there was supposed to be  add event that happened earlier, guess I'll artifically insert one." This should work fine, but it will require re-architecting the target management code a bit because there is currently no easy way to carry around this metadata.

phrack, 2015-08-04:
If you go to test this feature, don't try to use your old sessions (just delete the files in your sessions folder). The viewer doesn't try to fix broken sessions, the recorder just fixes sessions that would have otherwise been broken.

commit 053c77f4 (2015-08-04): Target events artificially added for targets used by events that existerd before recording started to close #150
closed 2015-08-04 by phrack