phrack/ShootOFF #151: Broke projection arena with backgrounds

cbdmaul, 2015-08-02:
Don't know if this is directly related to backgrounds or not.

Added savanna background.  Added plate rack and sized it down.  Went to move it and it jumped around and out of the arena.  When that happened, the plate rack and the background disappeared (Just returned to grey).

Now I can't change the background or add a target to it.

No exceptions thrown.

phrack, 2015-08-02:
Hah! I'll see if I can dupe it tomorrow.

phrack, 2015-08-03:
I have not duped the background issue, but I have duped the target issue. Interestingly, for any target on any canvas if you resize it with the mouse then try to move it with the mouse, the target runs away. This isn't a problem with other combinations:

- Resize with the mouse then move with the keyboard and it's fine
- Resize with the keyboard then move with the mouse and it's fine

It's only when you do both with the mouse that it runs way after a resize.

commit ab87ab27 (2015-08-03): Fixed coordinate updates for target movements to close #151
closed 2015-08-03 by phrack