phrack/ShootOFF #189: Add support for IP cameras

phrack, 2015-08-28:


cbdmaul, 2015-08-29:
is there rtsp support? rtsp is much more likely to be usable than mjpeg

phrack, 2015-08-29:
Even if this doesn't, there appear to be other options:

http://stackoverflow.com/questions/820888/java-rtsp-client-server-library

I don't really believe them, but someone claims that xuggler supports it, which is what we are using to record videos in debug mode and play them back in unit tests.

phrack, 2015-08-29:
This example works well for us: https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-drivers/driver-ipcam/src/examples/java/FoscamMjpegPushExample.java.

phrack, 2015-08-29:
Some dependencies listed here (http form apache);

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-drivers/driver-ipcam/pom.xml

Also need:

http://mvnrepository.com/artifact/com.github.sarxos/webcam-capture-driver-ipcam

commit 1792fb04 (2015-08-30): Added ipcam support to close #189
closed 2015-08-30 by phrack