phrack/ShootOFF #192: Implement DOS Shooting Gallery game training exercise

phrack, 2015-08-29:
http://www.dosgamesarchive.com/download/shooting-gallery/

https://www.youtube.com/watch?v=incPLdf712M

7III, 2016-02-19:
If you are looking for graphics for the shooting gallery exercise portion (not the "save the family" mini-game), here are a bunch of free vector images I found:

Rubber Duck: http://www.clipartbest.com/clipart-7Tapb6BTA
Target: https://pixabay.com/en/target-bull-s-eye-red-white-296483/
Squirrel: http://4vector.com/free-vector/squirrel-vector-4-130093
Pipe: http://clipart.me/free-vector-tobacco-pipe-10987
Wine: http://www.clipartbest.com/clipart-9ipe75BAT
Soda: http://www.clker.com/clipart-16792.html
Beer: http://www.clipartbest.com/clipart-ycoqMgzcE (looks a lot like the wine bottle)
Pinwheel: http://sweetclipart.com/pinwheel-toy-983
Apple: http://www.123freevectors.com/apple-graphic-free-vector/
Heart: http://www.clipartbest.com/free-heart-vector
Face: http://www.123freevectors.com/acid-smiley-face-vector-free/
Time Bonus: https://pixabay.com/en/clock-showing-nine-time-wall-clock-26104/
Ammo Bonus: https://pixabay.com/en/ammo-can-box-container-ammo-can-24553/
Coffee Mug: http://all-free-download.com/free-vector/download/coffee_mug_vector_47920.html
Candle: http://www.clipartbest.com/clipart-9c4oE95di

A couple of quick and dirty backgrounds:

Wood Shelves: http://wallpapercave.com/w/hZ78VC0
Pop-up shelves background: http://freedesignfile.com/90956-shiny-shelves-creative-background-vector-art-03/

(Apparently I had a lot of free time today)

7III, 2016-03-18:
Thanks to the flash game support (shot as click feature), this has become a bit of a moot point unless you are looking for exactly that game. There are a number of similar style flash games available. Thoughts on closing this enhancement?

phrack, 2016-03-18:
This game is actually still playable on most operating systems using DOSBox (me playing it is what inspired this issue). I think that's good enough to close this issue because why have a "remake" when you can play the real deal. Having said that, I'd still like to write up a small guide or make a video showing how to play this game using Shot to Click before closing this issue.

phrack, 2016-03-20:
Closing this in favor of Flash shooting galleries. The problem with DOSBox is that it handles mouse coordinates in its own special (DOS-specific) way, so there isn't a great way to get the coordinates right.

closed 2016-03-20 by phrack