phrack/ShootOFF #201: Add a split column to the shot timer

phrack, 2015-09-12:
Highlight (color?) reload and malfunction splits.

boom-headshot, 2015-09-16:
add a column to all training event types (except the ISSF 25m Standard Pistol) that calculates the elapsed time since the last shot was recorded, or the completion of the computer's audio input as applicable (i.e. beep or last listed target to shoot). The intent is to calculate the time between shots for competition shooters, or time to draw and engage a target.

boom-headshot, 2015-09-16:
add a colored and/or bolded horizontal line across all cells to indicate the start of one event and another (i.e. each input to draw). Intent is to differentiate strings of fire visually.

phrack, 2015-09-17:
The JavaFX TableView the shot timer uses doesn't support drawing custom lines. Might work to give different strings alternating colored backgrounds instead (e.g. string one has white background, string two light blue or gray, string three white, string four light blue or gray, etc.)

boom-headshot, 2015-09-17:
The proposed solution would work fine. At least two alternating colors would likely separate it adequately.

phrack, 2015-09-17:
This doesn't implement exactly what's specified (e.g. split based off beep) for two reasons:

1. That's a bigger architectural change than I am willing to make this late into the game for 3.3 (we've been working on this release for almost 6 weeks now and it already has too many big changes).

2. Places where you really want that already calculate their own split (e.g. timed holster drill).

Still, there is an overall split now and the split time is color coded in the table to let you know you had to reload or do a malfunction drill when you got that particular split.

commit eda2bc5f (2015-09-17): Added overall split time to shot timer with color coding for malfunctions and reloads to close #201
closed 2015-09-17 by phrack