phrack/ShootOFF #210: Plate rack + Record session = Stuck open session, no shot detection, shootoff won't close

cbdmaul, 2015-09-16:
Although it did crash eventually...

I can't give an exception because I am using JWS so that I can use the ps3eye.

phrack, 2015-09-16:
You should be able to see the exception in the Java Console, which you can open from the command line or the Java Control Panel. I get these same symptoms when I get #207 or a different but related xuggler exception.

cbdmaul, 2015-09-16:
It might be the same as #207 actually.  I didn't realize that was still open.  It is very hard for me to test with the broken ps3eye shot detection.  I would mark this invalid until I can test better.

phrack, 2015-09-16:
OK. I have a note on my phone to pack my camera so we can trade tomorrow.

closed 2015-09-16 by phrack