phrack/ShootOFF #222: Steel Target Audio Bug

boom-headshot, 2015-09-22:
When the plate rack or pepper popper are knocked down, the audio feedback for a hit on the remaining shape is not turned off. Also, there is no audio feedback for hits on the dueling tree

phrack, 2015-09-22:
There are a couple things here we need to tease out:

- The dueling tree is a separate issue (i.e. it should literally have its own issue). Just a quick "custom and courtesy" thing regarding bug reports: issues on issue tracks are typically supposed to be quite narrow. This is so that they can be closed in one or two commits to do a single bite sized thing. This makes the commit easy to review. We don't have enough devs right now to code review every single commit, but it's still a good thing to keep this habit for that purpose (I usually double check my own commits and do sometimes fine bugs I missed before committing just because the change is so easy to take in).
 - We just never added sound to the dueling tree but adding it isn't hard to ad. 

- The pepper popper behavior was intentional, but we also didn't fully understand pepper poppers at the time. This probably actually needs to versions of the target:
 - No "set screw" that falls for a hit anywhere and always has sound
- Yes "set screw" that only falls when you shoot it at the top but has the sound everywhere because you did shoot it (this is what we have now, but with the wrong shape)

- Is there a better metal clang sound? The one we have was a nice filler when it was added more than a year ago, but maybe there is something that doesn't sound like a pot being hit by a wooden spoon that would be better.

phrack, 2015-09-25:
This issue is now about doing three things:

- Use the new animation for the pepper popper.
- Use the new sound file instead of the current one.
- Have a version that animates every time and another that only animates for a head shot.

commit 8333a9bc (2015-09-26): Updated pepper popper graphics to close #222
closed 2015-09-26 by phrack