phrack/ShootOFF #227: Provide resizing anchor for mouse resizing of targets

cbdmaul, 2015-09-26:
Resizing targets is annoyingly difficult.  If I select a target, it should just put a resize anchor in the bottom right that I can grab.  Then when I deselect the target or the cursor leaves the window the anchor should disappear.

phrack, 2015-09-26:
Have fun with that one :p.

cbdmaul, 2015-09-26:
I know nothing about that code.

phrack, 2015-09-26:
Hah, well the reason this doesn't exist is because there is literally no support for this in Java. We'd either need an entirely custom control or some weird hack that uses images for the anchors. Once that works, resizing is still a bitch and is kind of glitchy. It'd probably need to be completely rewritten to make it so you can drag around the anchor images (assuming that hack is used) the dimensions of the shape are set based on the distances between the anchors. I really can't see it being at all easy to get everything aligned and working smoothly.

Whoever does this will wish they were making a new shot detection algorithm instead.

cbdmaul, 2015-09-26:
Well at the very least you can help me understand the problem more.

So you have code right now that moves the target that is clicked on and dragged, right? Obviously because that works.  Is that some kind of control that does it automatically or are you using the amount of movement of the mouse to determine the movement distance?

I wouldn't think it would be more complicated than:

1. Create a 5px x 5px image at the maximum x,y of the target.
2. When that image is clicked, in addition to moving the image itself, resize the target it is assigned to an equal number of x and y pixels to the distance moved of the mouse.

Is it really more complicated than that?

phrack, 2015-09-26:
It uses the movement of the mouse because there isn't another way to do it. This is all such a pain because there are no resizing primitives, thus they have to be made from scratch using mouse primitives that aren't really convenient for this use case. That would be seemingly easy except there are three different coordinate systems for a 2D pixel location in JavaFX and you have to convert between all three at different times and get it right every time otherwise weird things happen sometimes but not always. If you get all that working reasonably well there are annoying considerations to get right if you're going to do all of this extra work. Should you be able to shrink a target down to nothing? If not, when do you stop? What about moving the right most anchor to the left of what was the left most anchor. Is that possible? If so, what happens? If not, when do you stop the movement? Are any of these extra calculations different on a 4k display?

cbdmaul, 2016-03-29:
It now supports dragging from corners IF the target has a corner.

Only thing left is adding a resizing anchor to edges lacking corners

phrack, 2016-03-30:
A super easy way to do this would be add a square image to the target's group in the target's four corners. However, this would have the following downsides:

- While the target is selected shooting one of the resize anchors would register as a hit.
- The anchors would also scale with the target.

The right way to do that is, thereby avoiding this issue, is why I've been avoiding this issue.

cbdmaul, 2016-03-30:
I don't think having the resize anchor scale with the target is that big of an issue.  If you have some tiny target you don't want it to have a huge resize anchor (comparatively) anyway.

I'd only have the resize anchor show up when the target is selected also.

phrack, 2016-03-30:
I'll go ahead and try it this morning. I think we want to support ignoreHit as well. We don't have a target where we want to ignore a shot in some position, but it's easy to imagine that such a target exists.

phrack, 2016-03-30:
I ended up having to fix more coupling issues, but this ended up working quite well. It definitely would have been horribly painful when this was first open but after fixing the resizing bugs and doing a lot of decoupling work over the last two months, this got a lot easier to do.

I'll push after a clean up a few more small problems.

commit 30d76f84 (2016-03-30): Add resize anchors to selected targets to close #227
closed 2016-03-30 by phrack