phrack/ShootOFF #231: Add overlays for excessive motion, brightness, or other temporary disabled shot detection reasons

cbdmaul, 2015-09-29:
I think in general we should make an "Error" overlay for the feed and projection arena.

A good example is the one in gmail.  It puts a red bar in the top left with text that explains the issue.  Issue goes away = Bar goes away.  Simple as that.

I think "Needs calibration" could be converted to use this mechanism

Then during a bad shot detection situation (or no calibration), I can trigger that specific error, and untrigger it when it is corrected.

Caveat: Make color of error banner a bit lower brightness than everything else we're projecting.  We don't want it picked up as motion/brightness or it will cause a feedback loop of errors.

phrack, 2015-10-03:
I am sure you'll want to make some tweaks but I got the base code working to show diagnostic messages on canvases.

Example:

Label message = canvasManager.addDiagnosticMessage("This is a message", Color.RED);
canvasManager.removeDiagnosticMessage(message);

cbdmaul, 2015-10-03:
I was thinking that it should be in the top left like in gmail but having used this I see that having it in the middle is a bit more evident to the user which I think is okay in our case.

Do we want to add it to the projection arena also? If you are using a projector you're not exactly looking at the ShootOFF window examining it for errors.

phrack, 2015-10-03:
Would it fuck with shot detection on the arena? I don't look at the window when I am shooting, but I also tend to unfuck stuff when I am calibrating and I do look at the window while doing that.

Up to you.

cbdmaul, 2015-10-03:
If you could make it capable of fading out over, say, a second, it won't affect shot detection.

phrack, 2015-10-03:
The transparency could be lowered incrementally over a second until it's disappeared, then the control could be removed. Probably a ton of work.

phrack, 2015-10-03:
Alternatively, we could optionally play a chime of some kind to let people know there is a new error on they should look at their computer to read.

cbdmaul, 2015-10-03:
The problem with that is that the messages may only appear for a second.  Unless we want to wait some mandatory time before the user is alerted about errors? We could have it chime for the warning stays up for 5+ seconds...

phrack, 2015-10-03:
I probably need to just try it because I am having a hard time picture what the problem is at this point.

cbdmaul, 2015-10-03:
Even the dueling tree sets off the motion warning right now...which is fine in my opinion (If it wasn't the dueling tree but an equally sized object moving around, we'd want it to trigger), but if we're putting it up on the projection area or chiming, obviously there needs to be some kind of filter or delay.

phrack, 2015-10-03:
I see. Seems like the chime if it has been up for >5 seconds is the best idea because it would really highlight "HEY!! There is something important going on here and you should really stop and look at it!"

phrack, 2015-10-03:
If you agree with that I can add a chime if you take #238.

cbdmaul, 2015-10-03:
Agreed.

commit 5cd8fe58 (2015-10-03): Added chime for long displaying diagnostic messages to close #231
closed 2015-10-03 by phrack