phrack/ShootOFF #239: Exception selecting webcam + related webcam loading issue

cbdmaul, 2015-10-03:
ShootOFF does not get information off the Logitech camera right away.  When it starts with my built in camera selected (which works just fine) and I go to the menu to add the Logitech, the Logitech frame is blank.  I then add the Logitech and hit okay on the options menu and ShootOFF says it can't communicate with it.  If I then go into the menu again and delete and readd the Logitech, it throws the exception.

If I reload ShootOFF after having added the Logitech, ShootOFF works just fine with it.  So I have to add the Logitech and reopen ShootOFF every time I use a fresh ShootOFF.properties

[Thread-10] DEBUG com.github.sarxos.webcam.Webcam - Setting new resolution 640x480
[Thread-10] DEBUG com.github.sarxos.webcam.WebcamLock - Lock Webcam FaceTime HD Camera (Built-in) 0
[atomic-processor-1] INFO com.github.sarxos.webcam.ds.cgt.WebcamOpenTask - Opening webcam FaceTime HD Camera (Built-in) 0
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Opening webcam device FaceTime HD Camera (Built-in) 0
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Webcam device 0 starting session, size java.awt.Dimension[width=640,height=480]
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Webcam device session started
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Initialize buffer
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Webcam device com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice@3e8e553b is now open
[Thread-10] DEBUG com.github.sarxos.webcam.Webcam - Webcam is now open FaceTime HD Camera (Built-in) 0
[Thread-10] DEBUG com.shootoff.camera.ShotDetection.ShotDetectionManager - HIGH BRIGHTNESS - avgBrightPixels 199.56793587887967
[Thread-10] DEBUG com.shootoff.camera.ShotDetection.ShotDetectionManager - HIGH BRIGHTNESS - avgBrightPixels 199.6053388616757
[Thread-10] DEBUG com.shootoff.camera.ShotDetection.ShotDetectionManager - HIGH BRIGHTNESS - avgBrightPixels 199.63950393821284
[Thread-10] DEBUG com.shootoff.camera.ShotDetection.ShotDetectionManager - HIGH BRIGHTNESS - avgBrightPixels 199.67071140797938
[Thread-10] DEBUG com.shootoff.camera.ShotDetection.ShotDetectionManager - HIGH BRIGHTNESS - avgBrightPixels 199.69921730546133
[Thread-10] DEBUG com.shootoff.camera.ShotDetection.ShotDetectionManager - HIGH BRIGHTNESS - avgBrightPixels 199.7252555007181
[Thread-10] DEBUG com.shootoff.camera.ShotDetection.ShotDetectionManager - HIGH BRIGHTNESS - avgBrightPixels 199.74903961811555
[Thread-10] DEBUG com.shootoff.camera.ShotDetection.ShotDetectionManager - HIGH BRIGHTNESS - avgBrightPixels 199.77076478895773
[Thread-10] DEBUG com.shootoff.camera.ShotDetection.ShotDetectionManager - HIGH BRIGHTNESS - avgBrightPixels 199.79060925239673
[Thread-10] DEBUG com.shootoff.camera.ShotDetection.ShotDetectionManager - HIGH BRIGHTNESS - avgBrightPixels 199.80873581775455
[Thread-10] DEBUG com.shootoff.camera.ShotDetection.ShotDetectionManager - HIGH BRIGHTNESS - avgBrightPixels 199.8252932002548
[Thread-10] DEBUG com.shootoff.camera.ShotDetection.ShotDetectionManager - HIGH BRIGHTNESS - avgBrightPixels 199.84041724112234
[Thread-10] DEBUG com.shootoff.camera.ShotDetection.ShotDetectionManager - HIGH BRIGHTNESS - avgBrightPixels 199.85423202206127
[Thread-10] DEBUG com.shootoff.camera.ShotDetection.ShotDetectionManager - HIGH BRIGHTNESS - avgBrightPixels 199.86685088325464
[Thread-10] DEBUG com.shootoff.camera.ShotDetection.ShotDetectionManager - HIGH BRIGHTNESS - avgBrightPixels 199.87837735323788
[Thread-10] DEBUG com.shootoff.camera.ShotDetection.ShotDetectionManager - HIGH BRIGHTNESS - avgBrightPixels 199.8889059982748
[Thread-10] DEBUG com.shootoff.camera.ShotDetection.ShotDetectionManager - HIGH BRIGHTNESS - avgBrightPixels 199.89852319820451
[Thread-10] DEBUG com.shootoff.camera.ShotDetection.ShotDetectionManager - HIGH BRIGHTNESS - avgBrightPixels 199.90730785512514
[Thread-10] DEBUG com.shootoff.camera.ShotDetection.ShotDetectionManager - HIGH BRIGHTNESS - avgBrightPixels 199.91533204072772
[Thread-10] DEBUG com.shootoff.camera.ShotDetection.ShotDetectionManager - HIGH BRIGHTNESS - avgBrightPixels 199.9226615875918
[Thread-10] DEBUG com.shootoff.camera.ShotDetection.ShotDetectionManager - HIGH BRIGHTNESS - avgBrightPixels 199.92935662929366
[Thread-10] DEBUG com.shootoff.camera.ShotDetection.ShotDetectionManager - HIGH BRIGHTNESS - avgBrightPixels 199.93547209375836
[Thread-10] DEBUG com.shootoff.camera.ShotDetection.ShotDetectionManager - HIGH BRIGHTNESS - avgBrightPixels 199.9410581539032
[Thread-10] DEBUG com.shootoff.camera.ShotDetection.ShotDetectionManager - HIGH BRIGHTNESS - avgBrightPixels 199.94616063927
[Thread-10] DEBUG com.shootoff.camera.ShotDetection.ShotDetectionManager - HIGH BRIGHTNESS - avgBrightPixels 199.9508214120227
[Thread-10] DEBUG com.shootoff.camera.ShotDetection.ShotDetectionManager - HIGH BRIGHTNESS - avgBrightPixels 199.95507871039612
[JavaFX Application Thread] INFO com.github.sarxos.webcam.WebcamDiscoveryService - Discovery will not run - driver CompositeDriver does not support this feature
[Thread-10] DEBUG com.shootoff.camera.ShotDetection.ShotDetectionManager - HIGH BRIGHTNESS - avgBrightPixels 199.95896746241255
[Thread-48] DEBUG com.github.sarxos.webcam.Webcam - Setting new resolution 640x480
[Thread-48] DEBUG com.github.sarxos.webcam.WebcamLock - Lock Webcam Logitech HD Webcam C270 1
[atomic-processor-1] INFO com.github.sarxos.webcam.ds.cgt.WebcamOpenTask - Opening webcam Logitech HD Webcam C270 1
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Opening webcam device Logitech HD Webcam C270 1
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Webcam device 1 starting session, size java.awt.Dimension[width=640,height=480]
[Thread-10] DEBUG com.shootoff.camera.ShotDetection.ShotDetectionManager - HIGH BRIGHTNESS - avgBrightPixels 199.96251957244075
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Webcam device session started
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Initialize buffer
[JavaFX Application Thread] DEBUG com.github.sarxos.webcam.Webcam - Closing webcam FaceTime HD Camera (Built-in) 0
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Webcam device com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice@3c2c7512 is now open
[Thread-48] DEBUG com.github.sarxos.webcam.Webcam - Webcam is now open Logitech HD Webcam C270 1
[atomic-processor-1] INFO com.github.sarxos.webcam.ds.cgt.WebcamCloseTask - Closing FaceTime HD Camera (Built-in) 0
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Closing webcam device
[JavaFX Application Thread] DEBUG com.github.sarxos.webcam.WebcamLock - Unlock Webcam FaceTime HD Camera (Built-in) 0
[webcam-lock-[FaceTime HD Camera (Built-in) 0]] DEBUG com.github.sarxos.webcam.WebcamLock - Lock updater has been interrupted
[JavaFX Application Thread] DEBUG com.github.sarxos.webcam.Webcam - Webcam FaceTime HD Camera (Built-in) 0 has been closed
[Thread-48] DEBUG com.github.sarxos.webcam.Webcam - Closing webcam Logitech HD Webcam C270 1
[atomic-processor-1] INFO com.github.sarxos.webcam.ds.cgt.WebcamCloseTask - Closing Logitech HD Webcam C270 1
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Closing webcam device
[Thread-48] DEBUG com.github.sarxos.webcam.WebcamLock - Unlock Webcam Logitech HD Webcam C270 1
[webcam-lock-[Logitech HD Webcam C270 1]] DEBUG com.github.sarxos.webcam.WebcamLock - Lock updater has been interrupted
[Thread-48] DEBUG com.github.sarxos.webcam.Webcam - Webcam Logitech HD Webcam C270 1 has been closed
[JavaFX Application Thread] INFO com.github.sarxos.webcam.WebcamDiscoveryService - Discovery will not run - driver CompositeDriver does not support this feature
Exception in thread "JavaFX Application Thread" java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
	at java.util.ArrayList.rangeCheck(ArrayList.java:653)
	at java.util.ArrayList.get(ArrayList.java:429)
	at com.shootoff.gui.ImageCell.fetchWebcamImageView(ImageCell.java:150)
	at com.shootoff.gui.ImageCell.updateItem(ImageCell.java:113)
	at com.shootoff.gui.ImageCell.updateItem(ImageCell.java:1)
[... 73 lines cut ...]
[... cut ...]
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Closing webcam device
[frames-refresher-[1]] DEBUG com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Cancelling refresher
[JavaFX Application Thread] DEBUG com.github.sarxos.webcam.WebcamLock - Unlock Webcam Logitech HD Webcam C270 1
[webcam-lock-[Logitech HD Webcam C270 1]] DEBUG com.github.sarxos.webcam.WebcamLock - Lock updater has been interrupted
[JavaFX Application Thread] DEBUG com.github.sarxos.webcam.Webcam - Webcam Logitech HD Webcam C270 1 has been closed

cbdmaul, 2015-10-03:
I don't know if it helps but this is what it shows if the Logitech is added and I start shootoff:

[JavaFX Application Thread] DEBUG com.github.sarxos.webcam.Webcam - Setting new capture driver com.shootoff.camera.Camera$CompositeDriver@25988f38
[JavaFX Application Thread] INFO com.github.sarxos.webcam.WebcamDiscoveryService - Discovery will not run - driver CompositeDriver does not support this feature
[JavaFX Application Thread] INFO de.dfki.lt.tools.tokenizer.JTok - loading language resources for en from jtok/en
[JavaFX Application Thread] INFO de.dfki.lt.tools.tokenizer.JTok - loading language resources for en from jtok/en
[Thread-10] DEBUG com.github.sarxos.webcam.Webcam - Setting new resolution 640x480
[Thread-10] DEBUG com.github.sarxos.webcam.WebcamLock - Lock Webcam Logitech HD Webcam C270 1
[atomic-processor-1] INFO com.github.sarxos.webcam.ds.cgt.WebcamOpenTask - Opening webcam Logitech HD Webcam C270 1
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Opening webcam device Logitech HD Webcam C270 1
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Webcam device 1 starting session, size java.awt.Dimension[width=640,height=480]
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Webcam device session started
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Initialize buffer
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Webcam device com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice@127173c0 is now open
[Thread-10] DEBUG com.github.sarxos.webcam.Webcam - Webcam is now open Logitech HD Webcam C270 1

phrack, 2015-10-03:
Do a gradle eclipse and refresh (don't forget you may have to change compilation target after doing this) to get the new version of webcam-capture. I don't guarantee this fixes all of the problems, but it fixes the ones that:

1. I could duplicate
2. I know we can do something about

commit 4ab6e24e (2015-10-03): Copied mutable list to close #239
closed 2015-10-03 by phrack