phrack/ShootOFF #249: Add autocalibration for the projector arena

phrack, 2015-10-04:
Try to automatically do it, if it doesn't work automatically fall back to the manual purple box.

cbdmaul, 2015-10-13:
I have this working in OpenCV but there's more work to do.  It didn't seem as reliable as OpenIMAJ at first but I think that is just a matter of getting the configuration of opencv right.

I recommend we also implement remembering the monitor for the projection arena and/or automatically going to the secondary monitor and full screening.  It would be helpful to have access to a variable that tells me if the arena is full screened, so that auto calibration doesn't happen until the arena is correctly in place (Is there any case where someone would not want to full screen it? that could be an issue).

phrack, 2015-10-13:
You can check if the arena is full screen with arenaStage.isFullScreen:

https://github.com/phrack/ShootOFF/blob/master/src/main/java/com/shootoff/gui/controller/ProjectorArenaController.java#L137
https://docs.oracle.com/javase/8/javafx/api/javafx/stage/Stage.html#isFullScreen--

The only reason I can think of that you wouldn't want to be full screen is to move the arena to another screen, but then you have to recalibrate anyway.

Could just add a property to the ProjectorArenaController to return that assuming that's what you already have accecss to.

The screen API is fairly simple: https://docs.oracle.com/javase/8/javafx/api/javafx/stage/Stage.html. I've been going back on fourth on this. Should it automatically put the arena on the lowest resolution display since it's probably a projector or should it remember the screen which could but probably won't change order. Neither option is hard I just haven't decided which I think is better.

cbdmaul, 2015-11-05:
I'm still working on it.  Getting pixel perfect alignment is much more code than I was expecting, but it's coming along.  I'd rather go through the pain now and have it robust than implement some half measure.

Basic flow:

1. Display the chessboard
2. Find the chessboard corners
3. Measure the corner size
4. Estimate the whole chessboard and edge size
5. Use canny edge detection and hough line algorithm to find outline of chessboard (Verify against estimate)
6. De-distort and rotate the board so that it is rectangular and aligned

Steps 1-5 happen during calibration.  Step 6 happens every frame (For slow machines we could just use a best fit rectangle from step 5 and leave out step 6)

closed 2015-11-08 by phrack