phrack/ShootOFF #258: Stream debugger shows null value instead of brightness

phrack, 2015-10-27:
Also reports of it not showing the FPS

https://www.reddit.com/r/ShootOFF/comments/3puy9o/it_works_great_until_it_doesnt/cwepnw2?context=3

cbdmaul, 2015-10-28:
I noticed it's not showing up for me either (the FPS).  I haven't looked any closer.

We cap a minimum FPS for our filters and deduplication but we have no max.  Just a wild idea here but if our FPS is getting set to Inf or NaN or something big, we might have a really screwed up set of filters.

cbdmaul, 2015-10-30:
1. It is actually NOT the case that we calculate it.  We only calculate it for testcases.

Line 493: setFPS(webcam.get().getFPS()); 

2. We do cap max FPS at DEFAULT_FPS, which is 30

Line 507: webcamFPS = Math.min(newFPS, DEFAULT_FPS); 

My theory is that some cams are not giving an FPS and that might be part of our problem.  If we are using DEFAULT_FPS but the actual FPS is only 10, our filter is 3 times as long as we intend it to be.  Same for dedupes.

phrack, 2015-10-30:
Hmm, I see how that could cause our problems but it's weird to me that webcam-capture may not return an FPS. It calculates the FPS itself the same way we would, it's not trying to pull it from the camera.

phrack, 2015-11-01:
This issue was closed because the lighting label issue in the title bar is resolved, however the FPS issue is not. I added some log messages to hopefully help track down what is going on there. We can open an issue for that if the messages up drag out an issue.

commit 70e70618 (2015-11-01): Lighting condition is now optional for stream debugger and added additional camera related logging messages to close #258
closed 2015-11-01 by phrack