phrack/ShootOFF #262: Exception when setting projection arena screen

cbdmaul, 2015-11-02:
Related to 3ce9c66b929bd2dd940f4f8086c9be64335af933

My secondary monitor appears to be a negative offset (up and left) from my primary.  When I enabled the projection arena, it maximized on my primary display, not my secondary.  Moving it to the secondary caused the exception below.

shootoff.properties:

shootoff.arena.y=150.0
shootoff.arena.x=-1395.0

Exception that I get when I moved the arena from the primary to the secondary and full screened it:

Exception in thread "JavaFX Application Thread" java.lang.RuntimeException: java.lang.reflect.InvocationTargetException
	at javafx.fxml.FXMLLoader$MethodHandler.invoke(FXMLLoader.java:1770)
	at javafx.fxml.FXMLLoader$ControllerMethodEventHandler.handle(FXMLLoader.java:1653)
	at com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:86)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:238)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:191)
	at com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(CompositeEventDispatcher.java:59)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
	at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:54)
	at javafx.event.Event.fireEvent(Event.java:198)
	at javafx.scene.Scene$KeyHandler.process(Scene.java:3965)
	at javafx.scene.Scene$KeyHandler.access$1800(Scene.java:3911)
	at javafx.scene.Scene.impl_processKeyEvent(Scene.java:2040)
	at javafx.scene.Scene$ScenePeerListener.keyEvent(Scene.java:2502)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$KeyEventNotification.run(GlassViewEventHandler.java:197)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$KeyEventNotification.run(GlassViewEventHandler.java:147)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.lambda$handleKeyEvent$349(GlassViewEventHandler.java:228)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$$Lambda$392/108230258.get(Unknown Source)
	at com.sun.javafx.tk.quantum.QuantumToolkit.runWithoutRenderLock(QuantumToolkit.java:404)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.handleKeyEvent(GlassViewEventHandler.java:227)
	at com.sun.glass.ui.View.handleKeyEvent(View.java:546)
	at com.sun.glass.ui.View.notifyKey(View.java:956)
	at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at com.sun.glass.ui.win.WinApplication.lambda$null$145(WinApplication.java:101)
	at com.sun.glass.ui.win.WinApplication$$Lambda$36/2117255219.run(Unknown Source)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at sun.reflect.misc.Trampoline.invoke(MethodUtil.java:71)
	at sun.reflect.GeneratedMethodAccessor1.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at sun.reflect.misc.MethodUtil.invoke(MethodUtil.java:275)
	at javafx.fxml.FXMLLoader$MethodHandler.invoke(FXMLLoader.java:1765)
	... 31 more
Caused by: java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
	at java.util.ArrayList.rangeCheck(ArrayList.java:653)
	at java.util.ArrayList.get(ArrayList.java:429)
	at com.sun.javafx.collections.ObservableListWrapper.get(ObservableListWrapper.java:89)
	at com.shootoff.gui.controller.ProjectorArenaController.getArenaHomeScreen(ProjectorArenaController.java:102)
	at com.shootoff.gui.controller.ProjectorArenaController.canvasKeyPressed(ProjectorArenaController.java:236)
	... 41 more
Exception in thread "JavaFX Application Thread" java.lang.RuntimeException: java.lang.reflect.InvocationTargetException
	at javafx.fxml.FXMLLoader$MethodHandler.invoke(FXMLLoader.java:1770)
	at javafx.fxml.FXMLLoader$ControllerMethodEventHandler.handle(FXMLLoader.java:1653)
	at com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:86)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:238)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:191)
	at com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(CompositeEventDispatcher.java:59)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
	at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:54)
	at javafx.event.Event.fireEvent(Event.java:198)
	at javafx.scene.Scene$KeyHandler.process(Scene.java:3965)
	at javafx.scene.Scene$KeyHandler.access$1800(Scene.java:3911)
	at javafx.scene.Scene.impl_processKeyEvent(Scene.java:2040)
	at javafx.scene.Scene$ScenePeerListener.keyEvent(Scene.java:2502)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$KeyEventNotification.run(GlassViewEventHandler.java:197)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$KeyEventNotification.run(GlassViewEventHandler.java:147)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.lambda$handleKeyEvent$349(GlassViewEventHandler.java:228)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$$Lambda$392/108230258.get(Unknown Source)
	at com.sun.javafx.tk.quantum.QuantumToolkit.runWithoutRenderLock(QuantumToolkit.java:404)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.handleKeyEvent(GlassViewEventHandler.java:227)
	at com.sun.glass.ui.View.handleKeyEvent(View.java:546)
	at com.sun.glass.ui.View.notifyKey(View.java:956)
	at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at com.sun.glass.ui.win.WinApplication.lambda$null$145(WinApplication.java:101)
	at com.sun.glass.ui.win.WinApplication$$Lambda$36/2117255219.run(Unknown Source)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at sun.reflect.misc.Trampoline.invoke(MethodUtil.java:71)
	at sun.reflect.GeneratedMethodAccessor1.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at sun.reflect.misc.MethodUtil.invoke(MethodUtil.java:275)
	at javafx.fxml.FXMLLoader$MethodHandler.invoke(FXMLLoader.java:1765)
	... 31 more
Caused by: java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
	at java.util.ArrayList.rangeCheck(ArrayList.java:653)
	at java.util.ArrayList.get(ArrayList.java:429)
	at com.sun.javafx.collections.ObservableListWrapper.get(ObservableListWrapper.java:89)
	at com.shootoff.gui.controller.ProjectorArenaController.getArenaHomeScreen(ProjectorArenaController.java:102)
	at com.shootoff.gui.controller.ProjectorArenaController.canvasKeyPressed(ProjectorArenaController.java:236)
	... 41 more

phrack, 2015-11-02:
A negative x value makes sense if the screen is to the left instead of the right because the coordinates are relative to the primary screen.

This is the important part in our code:

com.shootoff.gui.controller.ProjectorArenaController.getArenaHomeScreen(ProjectorArenaController.java:102)

This is basically telling us that those coordinates aren't returning a screen for some reason. I would think the negative values wouldn't matter unless we need to offset them a little to keep from going off the screen in this line:

ObservableList<Screen> arenaHomeScreens = Screen.getScreensForRectangle(arenaStage.getX(), arenaStage.getY(), 1, 1);

I actually never tried it with the projector to the left so I guess I should try that.

phrack, 2015-11-02:
PS. Should also try with a 0 width and height (i.e. maybe we can get away with just the coords to not risk going off screen).

phrack, 2015-11-07:
I thought this would be easy to dupe, but I can't get negative coordinates with the Nvidea X windows driver, which both of my laptops use, regardless of the screen arrangement or primary/secondary configuration. I tried to search for the arenaHomeScreen using a 0 width and height, but I always get the primary screen if I do that.

After wasting a bunch of time on that dead end I tried to duplicate this bug on Mac (yea, so I was pretty desperate...). I could get negative coords no problem on Mac, so unlike with the Nvidea driver the coordinate system was what I expected. Regardless, I could not duplicate this bug. It behaved exactly as expected the entire time.

Basically, when the arena tries to auto place itself or when you hit the hotkey to full screen it will try to get the screen the arena is currently on first. For some reason that failed for you when you hit the hotkey to full screen (it said the arena isn't on any screen).

I have one hair brained idea that's probably a dead end: are your displays extended? Outside of this I have no ideas at the moment.

phrack, 2015-11-07:
Alternative idea: was the arena partially off screen (this also doesn't cause problems for me, but it might on Windows)?

cbdmaul, 2015-11-07:
1. Of course it is extended.  What's the alternative? I can't display an arena on cloned displays...

[image]

2. Don't know, but I can't seem to duplicate either.

3. I cleared my properties and tried it again

It said:

2015-11-06 20:30:42 [JavaFX Application Thread] DEBUG c.s.g.c.ProjectorArenaController - Two screens present
2015-11-06 20:30:42 [JavaFX Application Thread] DEBUG c.s.g.c.ProjectorArenaController - Found likely projector screen: resolution = 1920.0x1080.0, newX = -1920.0, newY = 0.0

But then it put it on my primary display (Not at -1920, 0)

So it still isn't working right.

cbdmaul, 2015-11-07:
Not moving it might be unrelated to the negative coordinates.  It didn't move it on to my projector either and that is to the right (positive coordinates), not left.

phrack, 2015-11-07:
How many monitors are in play? What's the resolution of each?

cbdmaul, 2015-11-07:
with monitor:

1920x1080 (at -1920, 0)
2880x1800 (primary)

with projector

2880x1800 (primary)
800x600 (at 2880, 0)

phrack, 2015-11-07:
Regarded extended, it's not called this on Linux. It's just "to the right", "to the left", or whatever direction. I assumed it was something else since this "extended" jargon seems to be a Windows thing. 

Anyway, the crash may be a red herring. Ignoring that, you seem to be experiencing something I noticed, which is that JavaFX treats the left most display as primary and not the one that is actually primary as far as the OS is concerned. 

For me with this arrangement even the main ShootOFF window would open on 2. It may be the case that it's best to just drop the primary display heuristic with two displays and use only "user moved to this display" and "smallest screen."

cbdmaul, 2015-11-07:
How about the one that the main window is NOT on?

phrack, 2015-11-07:
On Linux and Mac what would happen in this case (projector to the left) is the main ShootOFF window would open on the Projector initially. I'd grumble and move it to my actual primary display and open the arena. The arena would open on the projection then because it's not where the main ShootOFF window is. 

This is fine with me. Would this work on Windows based on what you've seen? I can also have it dump all screen data when that exception would otherwise occur on the future. 

To be clear, this is in place of the "not primary display" heuristic that is failing if there are exactly two screens.

cbdmaul, 2015-11-07:
yes, the main window always opens on my primary display for me.

phrack, 2015-11-07:
OK, I'll implement this slightly different heuristic tomorrow. ShootOFF opening on the left-most  display on *nix is an annoyance we'll probably just have to live with for now. There are worse GUI bugs. On the current Ubuntu LTS Swing windows open on an imaginary display if there are multiple monitors (you have to hit alt + F7 to manually move the Swing window to an actual screen).

phrack, 2015-11-07:
OK, arena placement is now relative to where the ShootOFF window is and there is much more detailed error reporting that will get auto reported if something unexpected happens.

closed 2015-11-07 by phrack

cbdmaul, 2015-11-07:
Still maximizing on my primary display.

2015-11-07 12:42:07 [JavaFX Application Thread] DEBUG c.s.g.c.ProjectorArenaController - Two screens present
2015-11-07 12:42:07 [JavaFX Application Thread] DEBUG c.s.g.c.ProjectorArenaController - Found likely projector screen: resolution = 1920.0x1080.0, newX = -1920.0, newY = 0.0

The newX and newY are correct but it isn't moving it to that display.

cbdmaul, 2015-11-07:
This code is working:
			arenaStage.setX(newX + 10);
			arenaStage.setY(newY + 10);

It puts it on the right display.  But the maximize maximizes it on the primary display.

phrack, 2015-11-07:
Huh, OK, so this is an oddity in setFullScreen on Windows. Wonder if there is a workaround. This definitely doesn't happen on Linux or Mac. I tested the crap out of it before committing.

phrack, 2015-11-07:
This is actually a good opportunity to test the fallback on Windows. Exit full screen because it was put on the wrong display. Move the arena to the correct one and now fullscreen again. If you close ShootOFF, re-open it, and start the arena now it should put it on the correct display from now on because it should have saved the correct display coords to shootoff.properties.

phrack, 2015-11-07:
NVM regarding failsafe message.

cbdmaul, 2015-11-07:
It would except that the full screen operation is maximizing it on the primary display.

Can you send an f11 to the window instead of using their setfullscreen function? Just an idea

phrack, 2015-11-07:
According to the docs (https://docs.oracle.com/javase/8/javafx/api/javafx/stage/Stage.html#fullScreenProperty):

"If the platform supports multiple screens an application can control which Screen the Stage will enter full-screen mode on by setting its position to be within the bounds of that Screen prior to entering full-screen mode.

However once in full-screen mode, Stage's x, y, width, and height variables will continue to represent the non-full-screen position and size of the window; the same for iconified, resizable, style, and opacity. If changes are made to any of these attributes while in full-screen mode, upon exiting full-screen mode the Stage will assume those attributes."

This seems to be what we are doing... Is any part of the stage hanging over onto the primary display after it's moved with:

    arenaStage.setX(newX + 10);
    arenaStage.setY(newY + 10);

cbdmaul, 2015-11-07:
No it's exactly where you'd expect it to be.

Maybe there is a delay on the move being executed you need to account for?

phrack, 2015-11-07:
The way to find out would be for you to comment out this line:

https://github.com/phrack/ShootOFF/blob/master/src/main/java/com/shootoff/gui/controller/ProjectorArenaController.java#L207

Then hit F11 yourself to see if does the right thing. If it does the right think, yea, maybe we need to insert a short sleep or something?

cbdmaul, 2015-11-07:
I did that already, that's how I know it's going to the right place

phrack, 2015-11-07:
So it seems the real question is: how long does the delay need to be? I am guessing we might as well use Thread.sleep because the required delay is probably short enough to not be a big deal. 

Try Thread.sleep(100 /*ms*/); and move up and down from there until it works while being small.

cbdmaul, 2015-11-07:
Ah, I have it figured out.  It isn't moving until the function returns.

Platform.runLater(() -> { toggleFullScreen(); });

fixes it

commit d5c3af5f (2015-11-07): Changed 2 screen behavior and additional error reporting to close #262