phrack/ShootOFF #265: Shot deduplication appears to be too aggressive

phrack, 2015-11-08:
https://www.reddit.com/r/ShootOFF/comments/3nrxzj/release_shootoff_34/cw9jwnd

cbdmaul, 2015-11-08:
It is actually set to 1/5 of a second - 200ms

I am testing the following changes in an effort to fix this:

1. Minimum deduplication period of 4 frames.  This is the minimum that passes our tests.
2. 1/8th a second deduplication period (If you do the math, it'll never be longer than 4 frames unless fps > 40)
3. Don't cap FPS at 30.  I don't see any reason to do this.  I'll test with ps3eye to verify it doesn't explode.
5.  Use euclidean distance for deduping with an acceptable threshold of 6.25 pixels distance (at 640x480, it is calculated dynamically).  This is the huge one I think will help people big time.

Regarding 5, the tests show our valid dupes have a max distance of like 3 pixels.  We're being wayyy too aggressive on distance.

cbdmaul, 2015-11-08:
I take that back slightly.  We do have a valid dupe in the tests at a distance of 8.1858.  I will try out a pixel distance of 12.5

cbdmaul, 2015-11-08:
I'm happy with my testing.  Rarely picked up dupes when intentionally having a bad trigger pull.  There's a separate issue where the dupe is shifted because of it being on the edge of a dark/bright region divide, and therefore is not detected as a dupe where it might have been before.  I don't want to abuse the dupe system because we have a different bug (Although that bug may ultimately be "acceptable").

Let me know if you want me to push it to master or what.

phrack, 2015-11-08:
It's safe to push at the moment. I don't have any pending changes that aren't up already (i.e. you should pull first).

commit 9dcb2245 (2015-11-08): Made changes to the deduplication processor to better handle dupe distances and webcam fps.  Hopefully to fix #265
closed 2015-11-08 by cbdmaul