phrack/ShootOFF #268: Fallback to manual calibration gracefully

cbdmaul, 2015-11-08:
Right now there's no indication that autocalibration is failing and that the user should do something.

phrack, 2015-11-08:
Are you taking this, otherwise, how do we know it failed?

cbdmaul, 2015-11-08:
Yes if you like my plan of attack:

1.1. Autocalibration starts immediately (As it does now)
1.2. Add indicator to feed that we are attempting auto calibration
1.3. Don't show manual bounds
2.1. If autocalibration succeeds, proceed as usual (remove message from 1.2)
2.2 If it fails after 10? seconds of trying, change message from 1.2 to "needs manual calibration" message of some sort and show bounds

I'd also like to have the following:
1. Disable calibration when arena is not maximized and show message saying to maximize it on projector feed
2. Reattempt autocalibration when full screen is resumed

cbdmaul, 2015-11-08:
I need a callback in shootoffcontroller when full screen is set or unset.  Best approach?

phrack, 2015-11-08:
The constructor for the arena controller that is NOT used for testing takes in the ShootOFFController now (but I don't think it stores it in a field). You can store it in a field and make a new method to call in ShootOFFController whenever toggleFullScreen is called in the arena controller.

commit 726ffe4f (2015-11-08): Closes #268
closed 2015-11-08 by cbdmaul