phrack/ShootOFF #271: Minor bugs associated with arena shot markers

cbdmaul, 2015-11-08:
Manual calibration is not accurate enough to prevent the shot markers themselves from being picked up as shots.  I suggest we only allow this for auto calibration, which is accurate enough to detect them as dupes.  

Second issue is that hitting reset when using shot markers immediately results in them being picked up again because the marker disappears.  I suggest we disable shot detection for a short period of time after reset is hit.

Side note: By the way, from testing this I know that the delay between us making an arena change and it being picked up by the camera is <1 frame (Shows up in next frame).  This is good to know for using autocalibration to mask out our changes to the arena.

cbdmaul, 2015-11-10:
I am dealing with the second issue by adding a function that temporarily suspends shot detection.  I am a bit worried about race conditions given the variety of states that shot detection and calibration can be in.  I don't want to overhaul the whole thing but I think we have to consider it in the future.  We need a state machine to manage everything.

closed 2015-11-10 by cbdmaul

phrack, 2015-11-10:
I found and fixed two race conditions this weekend using vmlens: https://marketplace.eclipse.org/content/vmlens. However, I never got it to work with a shot detection unit test because I kept running out of heap space trying to search the traces for race conditions. Probably need to ramp Eclipse's heap space to 8-12 GB. 

Seems like everything works well enough now days that we can get away without refactoring unless we find a major problem.

commit 152c4829 (2015-11-10): Closes #271