phrack/ShootOFF #274: All the tests that have newlines in the assertEquals fail on Windows

cbdmaul, 2015-11-08:
I've known this for a while but never bothered to tell you.  Windows uses CRLF and it is making all your string comparisons fail.

phrack, 2015-11-08:
Huh, I have no idea if there is a decent way to work around that issue...

phrack, 2015-11-08:
You should run all of the unit tests when in a clean state, then in Eclipse click the black arrow next to "Test Run History" and export a test report. Upload the test report to this post, otherwise it will be impossible for me to know precisely which are failing.

cbdmaul, 2015-11-08:
[java 20151108-153250.txt](https://github.com/phrack/ShootOFF/files/29559/java.20151108-153250.txt)

Also it looks like all the slashes in paths are the other way on Windows

phrack, 2015-11-08:
Please run the tests after pulling this commit. If they all pass, please close this. Otherwise, please post a new report. It's entirely possible I missed one or two.

cbdmaul, 2015-11-08:
org.junit.ComparisonFailure: expected:<sounds[/]beep.wav
Your score ...> but was:<sounds[\]beep.wav
Your score ...>
	at org.junit.Assert.assertEquals(Assert.java:115)
	at org.junit.Assert.assertEquals(Assert.java:144)
	at com.shootoff.plugins.TestBouncingTargets.testShootFourThenDontShoot(TestBouncingTargets.java:101)

org.junit.ComparisonFailure: expected:<score: 0[]
> but was:<score: 0[
]
>
	at org.junit.Assert.assertEquals(Assert.java:115)
	at org.junit.Assert.assertEquals(Assert.java:144)
	at com.shootoff.plugins.TestShootForScore.testReset(TestShootForScore.java:76)

org.junit.ComparisonFailure: expected:<sounds[/voice/shootoff-makeready.wav
sounds/]beep.wav
> but was:<sounds[\voice\shootoff-makeready.wav
sounds\]beep.wav
>
	at org.junit.Assert.assertEquals(Assert.java:115)
	at org.junit.Assert.assertEquals(Assert.java:144)
	at com.shootoff.plugins.TestISSFStandardPistol.testFullRound(TestISSFStandardPistol.java:104)

org.junit.ComparisonFailure: expected:<sounds[/voice/shootoff-makeready.wav
sounds/]beep.wav
> but was:<sounds[\voice\shootoff-makeready.wav
sounds\]beep.wav
>
	at org.junit.Assert.assertEquals(Assert.java:115)
	at org.junit.Assert.assertEquals(Assert.java:144)
	at com.shootoff.plugins.TestISSFStandardPistol.testFull150sThenReset(TestISSFStandardPistol.java:183)
	

org.junit.ComparisonFailure: expected:<sounds[/voice/shootoff-shoot.wav]> but was:<sounds[\voice\shootoff-shoot.wav
]>
	at org.junit.Assert.assertEquals(Assert.java:115)
	at org.junit.Assert.assertEquals(Assert.java:144)
	at com.shootoff.plugins.TestRandomShoot.testFiveSmallTarget(TestRandomShoot.java:93)

org.junit.ComparisonFailure: expected:<sounds[/voice/]shootoff-subtargets-...> but was:<sounds[\voice\]shootoff-subtargets-...>
	at org.junit.Assert.assertEquals(Assert.java:115)
	at org.junit.Assert.assertEquals(Assert.java:144)
	at com.shootoff.plugins.TestRandomShoot.testNoTarget(TestRandomShoot.java:65)

cbdmaul, 2015-11-08:
Github is fudging the comment, if you look at it raw it shows the problem

phrack, 2015-11-08:
Second try...

cbdmaul, 2015-11-08:
You applied the File.separatorChar fix backwards.  It is system dependent, you want to replace the system dependent slash with the one you expecting ("/").

Should be this: .replace(File.separatorChar, '/')

You also seemed to overwritten your own changes? I still have 5 other tests failing

phrack, 2015-11-08:
It may be best for you to fix the remaining ones based on the examples of my fixes. I am honestly guessing at what's wrong without being able to actually see it.

cbdmaul, 2015-11-08:
org.junit.ComparisonFailure: expected:<150s score: 5[
20s score: 0
10s score: 0
total score: 5
]
sounds\voice\shooto...> but was:<150s score: 5[
20s score: 0
10s score: 0
total score: 5]

i don't know what to do about these cases

phrack, 2015-11-08:
So in this case the .replace("\r\n", "\n") in getScoreString lines (92 and 96) might be the problem. The format string %n should actually be \r\n on Windows and \n on everything else.

cbdmaul, 2015-11-10:
I should have never mentioned that these freaking tests fail

commit 6b50bac6 (2015-11-10): Closes #274 hopefully
closed 2015-11-10 by cbdmaul