phrack/ShootOFF #275: Race condition on diagnostic messages in CameraManager

cbdmaul, 2015-11-08:
We use Platform.runLater to set the labels but this is not thread safe.  If we go to remove the label later but it hasn't been set yet by the other thread, we fail to remove it and the diagnostic message remains forever.

I know how to fix it.

cbdmaul, 2015-11-08:
Same problem with your mouseOnArenaLabel (and any others I don't know about).

phrack, 2015-11-08:
You can find all of the calls to addDiagnosticMessage by right clicking the different version in CanvasManager and clicking "Open Call Hierarchy".

cbdmaul, 2015-11-08:
I know I'm just lazy

phrack, 2015-11-08:
This is probably only a partial solution, but I suspect diagnosticFutures (CanvasManager line 87) should be a ConcurrentHashMap.

cbdmaul, 2015-11-08:
I think my solution works well enough to close this without addressing that.  We can reopen if we get a bug.

phrack, 2015-11-08:
I don't think this fix is sufficient and future proof unless you make the boolean fields volatile. I am quite confident that would do it.

commit 5284ecc8 (2015-11-08): diagnostic booleans -> volatile, closes #275
closed 2015-11-08 by cbdmaul