phrack/ShootOFF #287: Replace static references to feed height and width with non-static object references

cbdmaul, 2015-11-16:
We use static references to CameraManager.FEED_HEIGHT and FEED_WIDTH throughout the code.  These should be changed to references to the correct CameraManager's object members instead.  This will take some changes as the CameraManager is not passed to all the classes that use those static references.

This is to support future changes to ShootOFF.

cbdmaul, 2015-11-25:
1. Dedupe processor needs to be on a per-camera basis.  It is currently assigned to the config.  Isn't this a bug anyway? If you have two shots at the same place on two feeds, they'll be picked up as a dupe.

2. The GUI elements that reference the feed height and width for display purposes need to use some different global default, preferably user configurable, for how large to show it.  I would appreciate this as 640x480 is frickin tiny on a 4k display.  It follows that the target bounds would use this instead of the feed settings.

3. Rolling recorder should have its video settings separated from the feed settings.

4. Autocalibration should be given bounds

phrack, 2015-11-25:
Deduplicators should probably be static in the CanvasManager and be used at or towards the top of addShot. Agreed it's a bug for it to be universal and even in the Configuration at all.

cbdmaul, 2015-12-04:
I pushed a new branch, code in there now fixes the 1st problem.  I don't see that as a big enough bug to merge into master but I'll leave that to you.  Wasn't that much code and it passes tests.

phrack, 2015-12-04:
I am going to open a separate issue for the dedupe bug to make it so I can merge it early. Since this is a bug fix it fits within the current "bug fixes only" code freeze I have in place.

cbdmaul, 2015-12-05:
I pushed a new commit that makes it static and final in CameraManager.  Tests are passing.

phrack, 2015-12-05:
Is it mergable to close #300?

cbdmaul, 2015-12-05:
No, hold on, there's obviously something wrong with the commit if you look at the diff.

phrack, 2015-12-05:
I'll stand by until you say it's ready. I don't usually bother to look at the diff unless I am told the code is done.

cbdmaul, 2015-12-06:
Newest commit f8e11ed4da76125daa62788a57cc9d943fe823cf handles the outstanding issues and tests pass.  However, no other resolutions have been tested yet which needs to be done before this is ready for prime time.  There will undoubtedly be translation issues if the display res and feed res differ which has not yet been handled.

phrack, 2016-01-09:
Closed by pulll request #319

closed 2016-01-09 by phrack