phrack/ShootOFF #303: High FPS testcases are not deduplicated properly

cbdmaul, 2015-12-08:
This only affects testcases with high FPS (None of our current testcases).  The testcase harness for shot detection uses its own deduplicationprocessor, separately from the one inside cameramanager.  This means the threshold for deduplication is being updated in one instance but the shots are being deduplicated in a different instance.  So the threshold number of frames for deduplication never goes up when the FPS is high.

I have a fix for it in the other branch but it is modified too much to just pull in.  It's only a few LOC so I will get to it at some point.  Or we could just leave this open as a reminder since that branch will probably be merged before we add any other testcases anyway.

phrack, 2016-01-09:
Is this closeable due to https://github.com/phrack/ShootOFF/commit/c7ab24d088afed02609ee101dd9153dec04bdaa4 ?

cbdmaul, 2016-01-09:
yes

closed 2016-01-09 by phrack