phrack/ShootOFF #304: Exception if calibration option changed to cropfeed during calibration

cbdmaul, 2015-12-09:
Don't know how we didn't notice this sooner.

Exception in thread "Thread-11" java.util.NoSuchElementException: No value present
	at java.util.Optional.get(Optional.java:135)
	at com.shootoff.camera.CameraManager$Detector.streamCameraFrames(CameraManager.java:518)
	at com.shootoff.camera.CameraManager$Detector.run(CameraManager.java:416)
	at java.lang.Thread.run(Thread.java:745)

phrack, 2015-12-12:
What set of code is this from? It doesn't line up with master or the ResolutionFix branch, so I can only guess at the problem/cause.

cbdmaul, 2015-12-12:
Good point. I'll get a better exception for it

cbdmaul, 2015-12-12:
Nevermind, I thought I had only changed debugging lines in cameraManager but I changed something else.  I'm closing this.

closed 2015-12-12 by cbdmaul