phrack/ShootOFF #306: TrendNet IP Camera URL type.

dboling1, 2015-12-14:
Could you add the ability to use this type of URL to access a Trendnet ip camera.

http://user:camera@192.168.3.53:80/mjpeg.cgi?.mjpeg

phrack, 2015-12-14:
Are you getting a specific error message? When I try that URL I get the expected error message for the IP:PORT being wrong, not an inability to parse the URL.

dboling1, 2015-12-14:
The error I get is could not communicate with the IP at http://user:camera@192.168.3.53:80/mjpeg.cgi?.mjpeg

I get to the URL just fine using Firefox and VLC media player.

Explorer and chrome will not connect as they don't like the username and password format.

My system is win 7 ultimate 64 bit

phrack, 2015-12-14:
Yea, the problem is the username:password@address format. There are basically three options:

1. Look in the webcams manual or login page to see if there is a way to pass the credentials as get parameters (e.g. http://user:camera@192.168.3.53:80/mjpeg.cgi?.mjpeg&user=user&pwd=password).
2. Turn the password off if you can ensure your configuration only grants access to the camera on the local network.
3. We'd need to add support for a username and password field in the GUI. This is probably the best long term option, but it's also not high priority for us.

phrack, 2015-12-15:
I've refactored the initial comment and title to document the optimal solution (what I listed as 3 in the comment immediately above this one). No telling when we'd get to this because basically no one uses IP cameras, and so far you're the only one that might not be able to use the get request trick.

dboling1, 2015-12-15:
From my understanding all TrendNet ip cams use the http://username:password@ format and a lot of other ip cameras do also. There is no way to configure the TrendNet to format the URL to http://192.168.3.51:80/videostream.cgi?user=user&pwd=foscam1 like my foscam mjpeg camera does.
I disabled user access control where I didn't need a username or password, however the error I get is that the camera is being used by another application or I have more than 1 shootoff process running. (NOT). 

 I personally have 5 wireless IP cameras 2 Trendnet PTZ, 1 Foscam bullet mjpeg, 1 foscam bullet H264 HD, 1 Foscam  PTZ H264 HD. The price of these IP cameras have dropped to a point that they are very affordable and are really more desirable than a USB, serial or parallel camera since they are more versatile. I would guess you are going to start seeing a lot more of the cameras.

I would like to see an option to adjust the (local) camera within shootout as a lot of these cheap china made USB cameras have no way to adjust brightness, contrast, exposure, ect...  and rely on Microsoft's generic camera driver.

phrack, 2015-12-15:
OK, well, this issue documents what is needed to support alternate username/password entry. It's not hard but it's also not high on the priority list, thus not sure when we'll get to it. I suspect that most IP cameras aren't fast enough for consistent shot detection without RTSP, which isn't well supported generically software-wise.

We'll likely never support adjusting arbitrary USB cameras from the software because it would be a support nightmare considering we support 3 different operating systems. Supporting even camera capture relatively universally is a nightmare. However, we are working on finding a "high-end" camera to recommend where we will support these kinds of adjustments (mostly automatically -- i.e. having the software make the necessary adjustments for the user). The idea being, if you want to have shot detection that will definitely work well use any recommended camera, if you want a close to ideal experience use the "high-end" recommendation. This is an example of something above this issue on the priority list.

closed 2016-01-09 by phrack

dboling1, 2016-01-09:
Well, I guess it's up to you and the folks supporting this software, but being able to make it work for most platforms/cameras to me is important, your software is good, however you have a lot of work ahead to make your software to mainstream.

Thanks for the support,
Diane,
http://www.nracerts.org/

phrack, 2016-01-09:
You seem to misundestand why the issue was closed. The issue was closed because your request was fulfilled. You will be able to do what you asked in 3.6.

dboling1, 2016-01-09:
Your saying this issue was fixed or ignored ?

phrack, 2016-01-09:
Fixed. Look at the two notes above you comment a few minutes ago. They link to the code changes and a wiki update I made.

commit 50ad2496 (2016-01-09): Register IPCams using a username and password outside of URL to close #306