phrack/ShootOFF #315: PS3Eye

cybertech99, 2015-12-28:
Can you add PS3Eye support. I am not a Java developer but here is a forum post that may make it really straight forward to add. Thank you!

https://codelaboratories.com/forums/viewthread/238/

phrack, 2015-12-28:
This is going to be a really hard sell for a number of reasons:

1. We don't write the code to directly interact with cameras, we use an open-source third-party library that already has an issue open regarding the lack of PS3eye support: https://github.com/sarxos/webcam-capture/issues/401  (i.e. this is not our lane).
2. The PS3eye's Windows driver is not UVC (https://en.wikipedia.org/wiki/USB_video_device_class) compliant, thus it's more, extra work to introduce compatibility than you'd think (we'd have to write, test, and maintain Windows-only code to work with just this camera, which we don't have to do for any other case -- and the PS3eye isn't even that great of a camera). Unfortunately, this is what happens when manufacturers chose to ignore established standards for no good reason.
3. The PS3eye works well on Linux, thus there is an easy and free avenue already for people who do not want to buy the affordable recommended cameras. There is currently a gentleman working on a LiveCD customized for optimally running ShootOFF. That will make this option even easier if he releases it.

cbdmaul, 2015-12-28:
The PS3Eye does work on Windows in odd circumstances: You have to have the development kit Windows driver (Which isn't available anymore) and have to use the Java Web Start version of ShootOFF.

In any case, the PS3Eye is a terrible camera.

cybertech99, 2015-12-29:
Thanks went with Logitech C270

closed 2015-12-29 by cybertech99