phrack/ShootOFF #322: Noisy test log messages could bump into Travis-CI stdout buffer limit

phrack, 2016-01-09:
These messages should likely be trace:

Test: test testTargetSelection(com.shootoff.gui.TestCanvasManager) produced standard out/err: 2016-01-09 15:43:29 [JavaFX Application Thread] DEBUG com.shootoff.camera.CameraManager - setDetecting was true now false

Test: test testLogitechOutdoorGreen2(com.shootoff.camera.TestCameraManagerLogitech) produced standard out/err: 2016-01-09 15:45:55 [Test worker] DEBUG c.s.c.s.PixelClusterManager - Cluster 0: density 4 4 - 12.566370614359172 11 - 0.8753521870054244

phrack, 2016-01-09:
Assigned to cbdmaul because I am not sure if he still needs these at debug.

cbdmaul, 2016-01-09:
I fixed PixelClusterManager.  Will reduce the other one in the next commit.

commit b1e6e6da (2016-01-09): Closes #322, moves autocalibration video to correct location
closed 2016-01-09 by cbdmaul