phrack/ShootOFF #329: False green shots are detected when objects move around the projection arena.

phrack, 2016-01-11:
e.g. Bouncing Targets produces a large number of false green shots.

phrack, 2016-08-13:
Current status: masking didn't really worth because we didn't have a great solution to account for the delay between when the computer sends a frame to the projector and when that frame is actually shown. Our next idea is to use IR and filter visible light.

cbdmaul, 2016-08-15:
Adjusting the exposure down seems to be a very good fix for this.  I suggest closing this, opening "auto camera adjustment" as a new issue.  Maybe add an extra part to "Excessive motion" that says, "turn down your damn exposure"? Maybe only during exercises with motion?

phrack, 2016-08-15:
Have you tested changing the exposure from OpenCV on Windows? If so I'd like to see that it works on Linux too before the issue change.

cbdmaul, 2016-08-15:
Regardless, I think the warning is our best bet.  We are never going to fix this without camera adjustment, automatic or not.

phrack, 2016-08-15:
Totally agree the warning should be more actionable, just pointing out that there isn't enough info yet for me to be comfortable making an issue change in the short term.

cbdmaul, 2016-08-15:
It does work on Windows, tested on ocam/c270

phrack, 2016-08-15:
It works on Linux with the HD-3000, but it's not really clear what the values mean. I set both to zero and it was really weird, but it did do something.

I'll close this issue once I change the motion message to suggest lowering the webcam's exposure.

commit 05d16581 (2016-08-15): Make motion warning actionable to close #329
closed 2016-08-15 by phrack