phrack/ShootOFF #373: Application does not exit cleanly

edwardkort, 2016-02-11:
I am running Windows 10 and the latest JVM. If I invoke ShootOFF by double-clicking the jar file (which is associated with java binaries, the program runs fine. Exiting the program (by menu or close box) does not release the webcam and leaves the JVM running. The same thing happens if the program is invoked from a batch file with the command: java -jar ShootOFF.jar.

The only way to get the program to fully close is to add an exit command to the batch file.

Looking at the output in the command window (or in the Eclipse console) after closing the program shows a number of exceptions.

Cheers,
Edward Kort

phrack, 2016-02-11:
The exceptions on Windows are usually ignorable (e.g. not being able to delete a temporary file that doesn't matter.

What version are you using? We refactored some threading code that was causing this problem in 3.7. Would you mind running a nightly to see if we did indeed fix it?

http://dev.shootoffapp.com:9090/

The latest one may be a bit slow because the fix for the motion bug isn't optimized yet. If it ends up being too slow, I suggest the nightly from Feb 1. 

The console data actually generally isn't useful in this case. Useful data requires you to open VisualVM and get a thread dump. Don't use VisualVM for anything but a thread dump because this type of issue is often a Heisenbug (you try to monitor for it, it goes away -- seriously, attach VisualVM and suddenly it will close just fine and the bug will never happen if you try to instrument on startup).

edwardkort, 2016-02-11:
I downloaded the latest nightly build (02072016), unzipped it, and ran ShootOFF (by double-clicking the jar in an Explorer window). Exited the program through the File menu. The JVM was still listed in Task Manager.

phrack, 2016-02-11:
I see, this is a specific problem when File -> Exit is hit and not a problem when closing it by hitting the X. I can't dupe this problem with the X but I can dupe it reliably on Linux with File -> Exit. It seems that the onCloseRequest event isn't firing when close() is called on the ShootOFF stage by File -> Exit event handler:

https://github.com/phrack/ShootOFF/blob/master/src/main/java/com/shootoff/gui/controller/ShootOFFController.java#L1177

edwardkort, 2016-02-11:
I see the same under Windows 10. Hitting the close button exits cleanly.

phrack, 2016-02-13:
The specific problem is that the onCloseRequest event only fires if the window is closed externally, but we assumed it is closed even if you call close on the stage (NOT THE CASE!). This means that when we call close from File -> Exit the cleanup code in onCloseRequest is not called.

commit 69222199 (2016-02-13): Make ShootOFF close operations consistent to close #373
closed 2016-02-13 by phrack