phrack/ShootOFF #376: Essentially unuseable

tinyejp, 2016-02-15:
Several computers, several cameras tried.  Brand new laptop and Logitech c5615 1080p webcam, all the way down to an old netbook and old fashioned eyeball cam, and others in between.  Camera works great in all other applications.  FPS something like 5 or 6 in ShootOFF.  Shot detection only about 35%.
I will continue to fight with it.  I am thinking it is a limitation in Java.

phrack, 2016-02-15:
You'd have to state PC specs, particularly naming the processor, for "brand new laptop" to mean anything. There are plenty of "brand new laptops" that aren't very powerful. I am not at all surprised it wouldn't run well on an old netbook.

The HW requirements are relatively high at the moment and because of Java and because there hasn't been a concerted effort to optimize for performance (because still working on accuracy, but almost done). We are working on a native implementation of shot detection to reduce the HW requirements. That takes time though and the work is done here and there, not a super focused effort to get it done.

tinyejp, 2016-02-15:
I have to apologize. It turned out I had something eating up resources in the background.  Squared that away and all is well.  

I would have posted back sooner, but I've been giving my trigger finger a workout!  It was hard to put down once I found my error!!

Again, sorry.  Should have thought a little harder before opening a ticket.  I feel pretty foolish...

Shot detection is right around 90 percent, and I think I can improve that more with some environmental tweaks.  Projector arena is great!

closed 2016-02-15 by tinyejp

phrack, 2016-02-15:
Glad you figured it out! One way to increase accuracy of shot detection is to turn down exposure on the camera. 

https://github.com/phrack/ShootOFF/wiki/Camera-Setup-Tips

Many ShootOFF alternatives tweak camera settings to achieve optimal performance, but they can do this relatively easily because they only support one platform (accessing camera settings is _very_ platform specific). 

Anyway, there are valid performance complaints about ShootOFF so you're not completely off base, they just aren't as high of priority as other things at the moment.