phrack/ShootOFF #377: Entire target menu for projector arena not working

cbdmaul, 2016-02-18:
In master I am unable to add projector targets.  No errors, just nothing happens.

Courses add targets just fine.

phrack, 2016-02-18:
This is a total WTF. The code to add the targets was removed in this commit:

https://github.com/phrack/ShootOFF/commit/0a6e9216bc24bf99de1cc63719583bf23aa0899b#diff-a4f20fd1d632a7d0e0da6986506405d5

The merged commits are both yours, but you don't remove the code in either one and neither do I. As far as I can tell, the formating got super fucky in one of them (the code was still there though) and this made it different enough between the two that the merge decided to just nuke it.

closed 2016-02-18 by phrack

cbdmaul, 2016-02-18:
that is really strange.  it's not even a multiple line complicated merge.  just straight up deletes a line.

i blame the Illuminati

commit 4ac92828 (2016-02-18): Re-added code to add targets to arena from menu to close #377