phrack/ShootOFF #378: Shot markers being removed by reset causes false shots

cbdmaul, 2016-02-18:
Shot detection re-activated too quickly after reset likely culprit (There's no expectation that our shot detection would prevent these false positives).

phrack, 2016-02-18:
Is this a side effect of masking making things take longer?

cbdmaul, 2016-02-18:
I don't think so, I believe this has always happened to me.  The masking might fix this but masking will only happen if there is auto calibration so we should fix it anyway.

cbdmaul, 2016-03-01:
Having looked at this again, do we disable shot detection for reset? I can't find the code for it.

phrack, 2016-03-01:
https://github.com/phrack/ShootOFF/blob/master/src/main/java/com/shootoff/gui/controller/ShootOFFController.java#L967

I will test it.

cbdmaul, 2016-08-10:
This still happens.

cbdmaul, 2016-08-10:
Ugh.  Even extending it to a second isn't enough if there are shot detection processing delays like bright pixels.

phrack, 2016-09-09:
This seem to have dependencies on camera settings and such because I can't dupe it.

cbdmaul, 2016-09-25:
I haven't seen it either, let's close it

commit 97dd9fd1 (2016-03-01): Longer shot detection pause on reset to close #378
commit 6b0785e7 (2016-08-10): Fixes #378
closed 2016-09-25 by cbdmaul