phrack/ShootOFF #379: Preferences dialog "OK" button does nothing

slacey, 2016-02-18:
In the system prefs (File > Preferences), the OK button does not accept changes to the dialog and save/exit. The only way to get out of this screen is via Cancel and nothing is saved. 

3.6-final version (Windows 8.1 Dell XPS 13)

phrack, 2016-02-18:
There are two reasons I can think of that this could happen:

1. Your configuration could not be written to because it's in a location that requires more privileges than the account that ran ShootOFF (e.g. you put ShootOFF in C:\Program Files\ but didn't run ShootOFF as the administrator -- you can either set shootoff.home on the command line when running ShootOFF to put the settings and such in a different location or you can move all of ShootOFF out of this privileged location).
2. There is something it decided is invalid about the configuration and it's throwing a ConfigurationException as a result.

Please run ShootOFF from the command line and copy and paste the stdout output that occurs when you duplicate this problem (I am assuming you know what I am talking about since you have a GitHub account...). I can't duplicate it, thus it's something about the specific settings you are trying to save when you hit OK that is causing it.

slacey, 2016-02-18:
It is correct that it is in a privileged location. 
The fix is to move it from the privileged location to another dir and everything works, including log output.

Therefore, this fix should be closed with a notation in the 'installation' instructions, perhaps.

Thank you!

phrack, 2016-02-18:
There is actually an error message in a newer version when this happens. We just haven't released it yet.

closed 2016-02-18 by phrack