phrack/ShootOFF #385: Will not open on Windows 10

quinnjalan, 2016-02-22:
I downloaded and installed ShootOFF and Java 8. When I double click on ShootOFF.jar a window pops up and closes in a nano second. I tested on my work laptop (Win 8) and it opened without issue. However, I can't really leave it on there! Can ideas? I'll be happy to supply log files etc. Thanks, Alan

phrack, 2016-02-23:
@cbdmaul certainly knows more about Windows 10 than me and may have ideas. 

For starters, I suggest running ShootOFF-diagnostics.jar in the same directory that contains ShootOFF.jar. Wait about 15 seconds and a file called shootoff-diagnostics.log will appear. Please attach it to this thread by dragging and dropping it onto the comment field when you reply.

quinnjalan, 2016-02-23:
Thanks for getting back to me. I get the same results where a window pops
up and closes in a nano second and I do not get the expected
shootoff-diagnostics.log file in the shootoff folder. I tested on my Win 8
(work) laptop and I get a log (its blank) but at least it producing
something. Any more ideas? Cheers, Alan

phrack, 2016-02-23:
Huh, I am not surprised that you get the same flash, but the fact that you don't get a log even is disturbing.

Do you know how to use the command line?

cbdmaul, 2016-02-23:
I've used W10 for literally 30 seconds total but I have access to a W10 laptop.  I'll try to reproduce tonight.

quinnjalan, 2016-02-23:
phrack, yes on command line. What do you want me to do?

cbdmaul, thank on testing later

Cheers, Alan

phrack, 2016-02-23:
cd to the directory containing ShootOFF.jar. Run:

java -version

Then run:

java -jar ShootOFF.jar

Copy and paste any output here.

quinnjalan, 2016-02-23:
It opened via the Command Line. See attached image for my steps. Progress
is good!

phrack, 2016-02-23:
I don't think GitHub includes attachments from email replies. Can you reply on the website and drag and drop the images onto the comment box?

quinnjalan, 2016-02-23:
[image]

Try the above.

phrack, 2016-02-23:
OK, so it looks like it opens fine from the command line (although the shot timer on the right is smooshed). The question then is: why does it not open right when double clicked?

Two thoughts come to my mind:

1. Are there multiple versions of Java installed? If yes, an older version may be getting used, in which case it would fail when it realizes ShootOFF requires minimum Java 8.

2. The .jar extension may be associated with something else. You could try right clicking the .jar and looking at the Open With menu to see if it's associated with something other than Java.

quinnjalan, 2016-02-23:
See this video for what I'm doing and seeing: http://screencast.com/t/wqEz1p1tL1m
[ShootOff_Command_line_text.txt](https://github.com/phrack/ShootOFF/files/143273/ShootOff_Command_line_text.txt)

Also, I only have one version of Java installed Jave 8 Update 73

cbdmaul, 2016-02-23:
That error is irrelevant, it is failing to clean up a dll on closing ShootOFF.

Can you address phrack's 2nd idea?

quinnjalan, 2016-02-23:
#2 The .jar extension may be associated with something else. You could try right clicking the .jar and looking at the Open With menu to see if it's associated with something other than Java.

Made no difference. I changed the association away from and back to Java.

phrack, 2016-02-24:
There appears to be a solution linked to from here:

http://answers.microsoft.com/en-us/windows/forum/games_windows_10/cant-open-jar-file-using-windows-10/98f81ae5-0cbb-47f7-b7ff-49275e19e7a2?auth=1&rtAction=1441586909922

quinnjalan, 2016-02-24:
I ran the exe in the link and it asked if I wanted to associate jar file to
Java, I choose yes. The ShootOFF window popped up for a 1/2 nano second
longer than before and once again it close.

cbdmaul, 2016-02-24:
Was not able to reproduce.  Windows 10, java 8u73 x64, shootoff 3.6.  Ran just fine.

I suggest uninstalling java and reinstalling.  Make sure it is 8u73 x64.

quinnjalan, 2016-02-24:
Will test later on tonight when I get home. Thanks

phrack, 2016-02-27:
I am planning to close this tomorrow unless there is some compelling update.

closed 2016-02-28 by phrack