phrack/ShootOFF #393: Add GitHub contributor page to walk users through making a pull request starting from scratch

phrack, 2016-03-01:


ifly53e, 2016-03-01:
Requesting that the following question be addressed for this topic...  

1.  After I fork the repo in Github desktop, why can't I see the latest commit?  The date of the master fork seems to stay the same as the day I created my local copy.  I thought using the sync button at the top would keep everything up to date with the latest commit but, in my setup at least, the sync button does not seem to do anything.  For my pull requests, I have been downloading and creating an entire new branch for each one because it appeared that I did not have the latest commits.  This seems like an awful waste of disk space.  Am I supposed to do this?

2.  I now understand that I can't do a pull request until I have a commit of the code.  I also now understand that you can uncheck the boxes next to the files and those files will not be part of the commit.  I did not do this the first time and submitted a lot of files that I did not intend to submit.  What I don't get is why after I do a commit, when I press the "view on Github" link that I get a 404 error...that throws a lot of uncertainty into the equation when I can't seem to see the changes for my own repo...I
 just pressed the pull request button and hoped it all worked out....

3.  When am I supposed to do a pull request?  I did not want to break ShootOFF dev "protocol" by doing a pull request, but it seems that doing a pull request was the only way to get questions answered or help with coding.  This may not be the case anymore if reddit is going to be used for this...

4.  I also now know that after a pull request is closed by the devs that you can edit the files, fix any issues, and submit another pull request and everything stays in the same thread.  It is ok that only one file (the changed file) was uploaded for that commit because the other changed files will remain with the thread.  I am not sure if I just lucked out with this because I happened to be on the right branch in Github desktop, or if this is default behavior.

5.  I did not know that I was supposed to use a new class in Eclipse and modify ShootOffController to get things started either (this may all be different now that the module protocol is in place.)  I would have used the code formatter before my pull request had I known about it.  In the root directory of shootoff, there is a formatting.xml file.  You have to import it in Eclipse by going to File->Import... and at the bottom of the import selection dialog find import xml file.  After it is imported, select all the text (Ctrl-A) then do a (Shift-Ctrl-F) to format the code.

6.  I know we are not supposed to include dead code for a commit but what about leaving in troubleshooting code or leaving in debugging messages?  Maybe the reddit board takes care of this so that type of code does not have to be left in.

Thanks for creating this issue and putting all this together, Phrack.

phrack, 2016-03-02:
Notes:

1. When you do a fork you have a unique copy of the repository. The one at the phrack URL is still the "canonical" copy that all other repositories should update off of, but you won't see new commits in your copy unless you first do a pull with: git pull. This is because git is a distributed version control system instead of a centralized one. In a centralized system you have a "working copy" but no "repository" because the repository is on the server. In a distributed system everyone has a copy of the repository and the "working copy" is whatever is not commited. A "canonical" repository is the one everyone decides should be the repo that all commits are pushed to and that all the repositories update off of. In other words, the "canoical" repository should have all of the bleeding edge code that is stable enough to have in the master everyone considers the official master.

2. Committing only adds the changes to your local copy of the repository.  You have to "push" your changes to your remote copy repository on GitHub, but these are still not in the canonical repository (the one you cloned off of, this one) to make your changes accessible to everyone. If you commit it's only local, if you push it's in your remote copy of the repo (on GitHub), but to get those changes into the "canonical" repo you have to do a pull request. The thing to understand is that you basically have three repos because anything committed is in a repo with a distributed version control system (commit = local one, push = your remote one, and pull request = the conanical remote one). If you're updating you have to first merge from the canonical one to get the changes in your remote copy then pull to get them in your local copy.

3. You do a pull request after you have pushed your changes to your remote copy of the repo (your copy on GitHub) and are satisfied they are ready for review. A pull request basically says, "Hey, my copy of your repo has changes you might want, go ahead pull them yourself because I either can't or don't want to push them to your repository direclty." A pull request can also mean, "Hey, I want this stuff reviewed because I am confused and want help or just want early comments while I finish stuff up."

4. A pull request being closed basically means, "The changes you have in your remote copy that you want in the canonincal copy aren't done yet, please finish in your local copy with commits, push them to your remote copy, and finally make another pull request to ask me to look at and potentially include them in the canonical copy."

5. We can clarify these points easily by improving the docs.

6. Troubleshooting code should maybe be left in via a logger.isTraceEnabled block of code then you can enable it if you need it by enabling the trace log level in logback.xml. The shot detection code does this in a few places. Debug messages are fine to leave as long as you are using a logger at the appropriate log level (you can see logger.debug(....) code all over the place). 

4.

ifly53e, 2016-03-02:
This is awesome.  Thanks for your help.

commit fe011572 (2016-03-13): Added graphics for wiki page on git basics to close #393
closed 2016-03-13 by phrack