phrack/ShootOFF #394: Tests failing, java.lang.ExceptionInInitializerError

cbdmaul, 2016-03-01:
I did a gradle rebuild and project refresh.  Errors on testProjectorPlugin

java.lang.ExceptionInInitializerError
	at com.shootoff.plugins.ProjectorTrainingExerciseBase.<init>(ProjectorTrainingExerciseBase.java:51)
	at com.shootoff.plugins.SteelContest.<init>(SteelContest.java:40)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:422)
	at java.lang.Class.newInstance(Class.java:442)
	at com.shootoff.plugins.engine.Plugin$PluginSettingsXMLHandler.startElement(Plugin.java:142)
	at com.sun.org.apache.xerces.internal.parsers.AbstractSAXParser.startElement(AbstractSAXParser.java:509)
	at com.sun.org.apache.xerces.internal.parsers.AbstractXMLDocumentParser.emptyElement(AbstractXMLDocumentParser.java:182)
	at com.sun.org.apache.xerces.internal.impl.dtd.XMLDTDValidator.emptyElement(XMLDTDValidator.java:766)
	at com.sun.org.apache.xerces.internal.impl.XMLDocumentFragmentScannerImpl.scanStartElement(XMLDocumentFragmentScannerImpl.java:1343)
	at com.sun.org.apache.xerces.internal.impl.XMLDocumentScannerImpl$ContentDriver.scanRootElementHook(XMLDocumentScannerImpl.java:1292)
	at com.sun.org.apache.xerces.internal.impl.XMLDocumentFragmentScannerImpl$FragmentContentDriver.next(XMLDocumentFragmentScannerImpl.java:3138)
	at com.sun.org.apache.xerces.internal.impl.XMLDocumentScannerImpl$PrologDriver.next(XMLDocumentScannerImpl.java:880)
	at com.sun.org.apache.xerces.internal.impl.XMLDocumentScannerImpl.next(XMLDocumentScannerImpl.java:606)
	at com.sun.org.apache.xerces.internal.impl.XMLDocumentFragmentScannerImpl.scanDocument(XMLDocumentFragmentScannerImpl.java:510)
	at com.sun.org.apache.xerces.internal.parsers.XML11Configuration.parse(XML11Configuration.java:848)
	at com.sun.org.apache.xerces.internal.parsers.XML11Configuration.parse(XML11Configuration.java:777)
	at com.sun.org.apache.xerces.internal.parsers.XMLParser.parse(XMLParser.java:141)
	at com.sun.org.apache.xerces.internal.parsers.AbstractSAXParser.parse(AbstractSAXParser.java:1213)
	at com.sun.org.apache.xerces.internal.jaxp.SAXParserImpl$JAXPSAXParser.parse(SAXParserImpl.java:649)
	at com.sun.org.apache.xerces.internal.jaxp.SAXParserImpl.parse(SAXParserImpl.java:333)
	at javax.xml.parsers.SAXParser.parse(SAXParser.java:195)
	at com.shootoff.plugins.engine.Plugin.<init>(Plugin.java:79)
	at com.shootoff.plugins.engine.TestPlugin.testProjectorPlugin(TestPlugin.java:35)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at org.junit.runners.model.FrameworkMethod$1.runReflectiveCall(FrameworkMethod.java:50)
	at org.junit.internal.runners.model.ReflectiveCallable.run(ReflectiveCallable.java:12)
	at org.junit.runners.model.FrameworkMethod.invokeExplosively(FrameworkMethod.java:47)
	at org.junit.internal.runners.statements.InvokeMethod.evaluate(InvokeMethod.java:17)
	at org.junit.internal.runners.statements.RunBefores.evaluate(RunBefores.java:26)
	at org.junit.runners.ParentRunner.runLeaf(ParentRunner.java:325)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:78)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:57)
	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:290)
	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:71)
	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:288)
	at org.junit.runners.ParentRunner.access$000(ParentRunner.java:58)
	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:268)
	at org.junit.runners.ParentRunner.run(ParentRunner.java:363)
	at org.eclipse.jdt.internal.junit4.runner.JUnit4TestReference.run(JUnit4TestReference.java:50)
	at org.eclipse.jdt.internal.junit.runner.TestExecution.run(TestExecution.java:38)
	at org.eclipse.jdt.internal.junit.runner.RemoteTestRunner.runTests(RemoteTestRunner.java:459)
	at org.eclipse.jdt.internal.junit.runner.RemoteTestRunner.runTests(RemoteTestRunner.java:675)
	at org.eclipse.jdt.internal.junit.runner.RemoteTestRunner.run(RemoteTestRunner.java:382)
	at org.eclipse.jdt.internal.junit.runner.RemoteTestRunner.main(RemoteTestRunner.java:192)
Caused by: java.lang.IllegalStateException: Toolkit not initialized
	at com.sun.javafx.application.PlatformImpl.runLater(PlatformImpl.java:273)
	at com.sun.javafx.application.PlatformImpl.runLater(PlatformImpl.java:268)
	at com.sun.javafx.application.PlatformImpl.setPlatformUserAgentStylesheet(PlatformImpl.java:550)
	at com.sun.javafx.application.PlatformImpl.setDefaultPlatformUserAgentStylesheet(PlatformImpl.java:512)
	at javafx.scene.control.Control.<clinit>(Control.java:87)
	... 50 more

phrack, 2016-03-01:
These line numbers don't even make sense. You're getting an exception for a field declaration.

phrack, 2016-03-01:
Do you have multiple JDK's installed? It looks like for some reason you aren't using the Oracle JDK anymore. Closing this due to that observation, but discussion can continue.

cbdmaul, 2016-03-01:
what is that observation based on? i only have one jdk installed, oracle 1.8.0_51, although it is slightly out of date

phrack, 2016-03-02:
My observation is that collectively this exception and the dupe are claiming that javafx.scene.control.Label doesn't exist. It's a JDK class in 1.8 that is definitely in Oracle's JDK but not necessarily OpenJDK unless OpenJFX is installed.

cbdmaul, 2016-03-02:
Okay...but I am running oracle's jdk.  I do not have openjdk.  It's difficult to even get openjdk for Windows so I don't know why you'd think I was running it.

cbdmaul, 2016-03-02:
I updated to the latest JDK and it is still happening.

phrack, 2016-03-02:
I was guessing based on the fact that it thought a standard JFX class isn't present. I have no idea why this is happening on your system but it isn't a Windows issue because it's not happening in my Windows VM. It's something annoying with one of the build tools as usual.

phrack, 2016-03-02:
PS. Sometimes removing and re-adding the JDK fixes this type of issue for me. Not guaranteed to work and I have no idea what causes it.

cbdmaul, 2016-03-02:
I tried that.  And updating the JDK.  And updating Eclipse.  And cleaning the project directory.  And rebuilding the eclipse project.  I still get this error and it happens occasionally when I am running ShootOFF main rather than a test as well.

phrack, 2016-03-02:
The fact that it happens randomly when running ShootOFF main pushes this to total WTF territory.

phrack, 2016-03-02:
This SO question clarifies that jfxrt is in fact the JAR file you need with the Oracle JDK:

http://stackoverflow.com/questions/27255278/junit-tests-fail-if-run-as-part-of-a-test-suite

If you expand the JDK library in your build path in Eclipse you'll see it. The only reason I am aware of that there could be a mismatch in Eclipse is that the run configuration does not have the same class path. You could right click the project, click Run As -> Run Configurations then look at the JDK tab in the run configuration to see if it's fucky somehow. There should be a run profile for running from main and one for unit tests.

cbdmaul, 2016-03-02:
It's consistent across all of those.  jre1.8.0_74, which is the workspace jre.

The JRE system library set in the java build path is jre1.8.0_74.  there are no others present.

The compliance level is set to 1.8.

The jfxrt.jar is listed in the library.

It doesn't have to do with this does it? https://bugs.openjdk.java.net/browse/JDK-8123361

phrack, 2016-03-02:
The point of that discussion is that you need to initialize JFX before using one of it's components. That's why we have this rule: https://github.com/phrack/ShootOFF/blob/master/src/test/java/com/shootoff/plugins/TestSteelChallenge.java#L34. I could believe that something fucky is happening with that rule, but it wouldn't explain why you get the problem randomly when running from main.

phrack, 2016-03-02:
I updated to the same JDK today and I still can't dupe this even if I run just the steel test. I will test the other changes real quick and take a camera with me to the university so that I can at least try to dupe the pattern one.

cbdmaul, 2016-03-02:
I can try to confirm that it happens with main tonight.  I definitely saw the error while I was running main but maybe I had a test running in another instance at the same time or something.

Is that rule in all the tests that use javafx? because I have this error on like 20 different tests.

phrack, 2016-03-02:
It is, you'll find it all over the place because most non-shot detection tests have some JavaFX dependency.

cbdmaul, 2016-03-02:
HA IT IS A BUG

Needed to add the rule line to TestPlugin.  Now they all work fine.

(In your face)

closed 2016-03-02 by cbdmaul

phrack, 2016-03-02:
Oh man, I forgot that even existed.

commit 9efa25c9 (2016-03-02): Fixes #394