phrack/ShootOFF #396: Projection arena not finding second screen, not allowing maximizing on proper screen

cbdmaul, 2016-03-02:
Arena is maximizing on primary display (Same display as ShootOFF is running on).

Then If i manually move the arena to the right screen and hit f11, I get this error and it doesn't maximize.

2016-03-01 19:17:00 [JavaFX Application Thread] ERROR c.s.g.c.ProjectorArenaController - Didn't find screen for stage with title Projector Arena at (2630.000000, 212.666672). Existing screens: 

(w = 1920.000000, h = 1200.000000, dpi = 147.000000)
(w = 683.000000, h = 512.000000, dpi = 46.000000)

phrack, 2016-03-02:
This happens intermittently for me as well. It's like JFX doesn't consistently assign the screens from JVM instance to JVM instance. Not sure we can do anything about it since it seems random.

cbdmaul, 2016-03-02:
It's happening every time with my projector.  Weirdly didn't happen with my external display.  But I can't test some bug fixes I made without the projector.

phrack, 2016-03-02:
It never happens with my projector but it happens about 1 in every 4 times with an external display I have at the university. The auto-placement code is here:

https://github.com/phrack/ShootOFF/blob/master/src/main/java/com/shootoff/gui/controller/ProjectorArenaController.java#L148

Right now the priority is supposed to be:

- Show on the screen that belongs to the saved coord (JFX treats all screen as one big coordinate space).
- Show on the smallest screen if there is not a saved coord.
- If there is no saved coord and no smallest screen, just show on the primary display.

I don't have a problematic external display with me, but my best guess is that JFX doesn't get the screens consistently which means the overall coordinate space can change depending on how JFX randomly assigns the screens. This would mean that saving the coordinates as we do doesn't really work because your saved coords could not exist if you happened to not get the same display order in this instance of the JVM. If that is true, it could be best to just not save the display coord and just not do what is the currently the highest priority. This would be a shame because the smallest display is not necessarily the projector (the university has projectors that are higher than a laptop display). The entire reason behind saving the coord was so that you could say, "OK, the heuristic is wrong, but you corrected it so we'll remember what the correction was."

cbdmaul, 2016-03-02:
I have yet to have it work with my projector.  Has something changed? It always worked before.

phrack, 2016-03-02:
This code hasn't been touched since it was committed.

One potential small fix might be to alther this code:

https://github.com/phrack/ShootOFF/blob/master/src/main/java/com/shootoff/gui/controller/ProjectorArenaController.java#L357

Maybe move this line https://github.com/phrack/ShootOFF/blob/master/src/main/java/com/shootoff/gui/controller/ProjectorArenaController.java#L374 above 357 so that it will full screen either way, but it won't try to save the settings unless it could calculate the screen. This would be enough to allow you to actually use ShootOFF, but it still wouldn't find your projector automatically.

cbdmaul, 2016-03-02:
This doesn't make any sense.  I never had a problem with this before and now it's happening constantly.  I have not been able to use the arena at all since I pulled from 3.8

phrack, 2016-03-02:
It's been pretty random as far as I've experienced with my university display which is why I've never been around to find a work around. I think the best option for now is to move line 374 in the ProjectorArenaController just above line 357 so that ShootOFF can at least be made to function (right now it will not fullscreen if it can't find the screen to save, this small change would make it fullscreen even if it can't find the screen so that the arena will at least work for that session)..

phrack, 2016-03-02:
Give me a thumbs up in the next 10 minutes and I'll make that line move if you want. Again, this doesn't fix the auto-placement problem where the root cause is unknown, but it would at least provide a reasonable and easy way to make ShootOFF work with manual placement.

cbdmaul, 2016-03-02:
I'd say go ahead and make the change.  I'm too burned out to look at a
computer right now.

phrack, 2016-03-02:
Done. I didn't close this but this actually probably is a good enough workaround to close.

phrack, 2016-03-02:
PS. I suggest you either commit your fixes tonight and I can triage tomorrow or try to commit before work tomorrow. My concern is that my upload rate is shit at home so I'd really like to release at work if possible since the automatic release code has never actually done the full thing end to end yet (i.e. if it's broken I'd rather have faster turn arounds in fixing it with blazing fast Internet than having 30+ minute iterations).

cbdmaul, 2016-03-02:
This does allow me to move the arena to the correct screen and maximize it now.

I'm still very suspicious that this used to work perfectly for me and now does not work automatically like it used to.

phrack, 2016-03-02:
It's not the merge because even the blame data says we haven't touched that code since December and that was just to format, it hasn't changed functionally since early November:

https://github.com/phrack/ShootOFF/blame/master/src/main/java/com/shootoff/gui/controller/ProjectorArenaController.java

I am more inclined to believe something changed in a JDK update:

http://www.oracle.com/technetwork/java/javase/8u74-relnotes-2879024.html

cbdmaul, 2016-03-02:
That'd be unfortunate but seems like the only likely culprit at this point unless there is other code that is involved that we don't realize.

phrack, 2016-03-02:
You're on a roll! Will it be 3 for 3?

cbdmaul, 2016-03-02:
I'm looking at it now.  My only thought is, maybe something has changed but maybe there's something we can do about it.

phrack, 2016-03-02:
It behaved the same way on my university machine (with latest JDK) as it did before. There might be a better strategy that is being used, otherwise I dunno.

cbdmaul, 2016-03-02:
2016-03-02 18:06:01 [JavaFX Application Thread] DEBUG c.s.g.c.ProjectorArenaController - Found likely projector screen: resolution = 853.0x683.0, newX = 1920.0, newY = 0.0

Those numbers just...don't make any sense.  That's not near my projector resolution (Even including hypothetical taskbars that aren't there) and the coordinates are not near where that display is.  Plus neither are 1920xanything so I don't know where that's coming from.

phrack, 2016-03-02:
I was trying to track down this code in OpenJDK months ago to try to understand why some weirdness happens. It dead ends at platform specific native code, thus the outcome won't necessarily even be consistent from platform to platform.

cbdmaul, 2016-03-02:
bounds and visual bounds when run on 2880x1800 (primary) and 1280x1024 (projector) displays

2016-03-02 18:12:49 [JavaFX Application Thread] DEBUG c.s.g.c.ProjectorArenaController - screen Rectangle2D [minX = 0.0, minY=0.0, maxX=1920.0, maxY=1200.0, width=1920.0, height=1200.0] Rectangle2D [minX = 0.0, minY=0.0, maxX=1920.0, maxY=1175.0, width=1920.0, height=1175.0]
2016-03-02 18:12:49 [JavaFX Application Thread] DEBUG c.s.g.c.ProjectorArenaController - screen Rectangle2D [minX = 1920.0, minY=0.0, maxX=2773.0, maxY=683.0, width=853.0, height=683.0] Rectangle2D [minX = 1920.0, minY=0.0, maxX=2773.0, maxY=683.0, width=853.0, height=683.0]

So it looks like a javafx bug.  I couldn't find anything online about it.

phrack, 2016-03-02:
Do we think this issue is sufficiently mitigated to close given the commits from last night? If so, close away!

cbdmaul, 2016-03-02:
Might be better to keep it open so that we can attach it to a javafx inquiry, plus send people to this page upon inevitable complaints.

phrack, 2016-03-02:
Good point. I am unassinging the milestone as an acknowldegement that I am accepting this bug will live in the 3.7 release.

cbdmaul, 2016-03-02:
Followup:  It appears to all be wrong by 2/3s.  Meaning everything is divided by 2/3.  It looks like multiplying it all by 1.5 would make the resolutions (and therefore coordinates) correct.

Why? I don't know.

cbdmaul, 2016-03-02:
I know what this is.

http://stackoverflow.com/questions/9383351/how-do-i-get-the-retina-display-resolution

http://stackoverflow.com/questions/35555338/get-screen-scale-in-javafx-using-only-public-api

(Same issue) https://github.com/codecentric/javafxsvg/pull/3

phrack, 2016-03-02:
WTF!!! You shouldn't have to use com.sun classes to get the right answer! Those are supposed to be internal only.

cbdmaul, 2016-03-02:
Even using the internal API I get 2.0f, which isn't right.

phrack, 2016-03-02:
Looks like the Screen class gives it to you:

https://docs.oracle.com/javase/8/javafx/api/javafx/stage/Screen.html#getDpi--

cbdmaul, 2016-03-03:
Yes, but how do you go from DPI to knowing it was scaled down 1/3?

DPIs are 147 and 63

Rectangle2D [minX = 0.0, minY=0.0, maxX=1920.0, maxY=1175.0, width=1920.0, height=1175.0] 147.0 2.0 2.0
2016-03-02 18:46:34 [JavaFX Application Thread] DEBUG c.s.g.c.ProjectorArenaController - screen Rectangle2D [minX = -1280.0, minY=0.0, maxX=0.0, maxY=720.0, width=1280.0, height=720.0] Rectangle2D [minX = -1280.0, minY=0.0, maxX=0.0, maxY=720.0, width=1280.0, height=720.0] 63.0 2.0 2.0

I do know that they aim for 96 DPI, so..

147/96 = 1.53125
96/63 = 1.52380

...What do we do, round it?  This is the most hackish thing I've considered in a while.

phrack, 2016-03-03:
It's software man, you kill a pig and rub the blood on your screen and hope it works. Seriously though, how solid of a standard is 96 DPI?

cbdmaul, 2016-03-03:
Oh like there would be one standard.  That'd be crazy talk.

http://news.kynosarges.org/2015/08/24/javafx-gui-scaling-at-120-144-dpi/

96 is 100%, so it is default.  Except when the DPI is 120, and then it isn't the default.

cbdmaul, 2016-03-03:
https://bugs.openjdk.java.net/browse/JDK-8090569

cbdmaul, 2016-03-03:
I see what the cause is.  I do have Windows set to scale to 150%, although only on my primary display.

It appears that jfx has no support for these conditions at all.  We are better off doing nothing and complaining bitterly about it.

I suggest keeping this open as per our discussion above.

cbdmaul, 2016-07-06:
In an odd turn of events, this appears to be working for me now.  But I upgraded to Windows 10 and updated the JRE at the same time, so I don't know that it is actually any different for Windows 8.1 at the current time.

phrack, 2016-07-06:
Given that the DPI issues remain I am inclined to believe it was the Windows 10 upgrade. If you have a Windows 8.1 system to test, the results should prove interesting.

closed 2016-08-13 by cbdmaul

phrack, 2016-08-13:
I can confirm this fix didn't break anything on a system where DPI doesn't come into play. Looks like this is finally over!

commit 395a2537 (2016-08-13): Fixes #396! I think! Maybe?
commit b58700ed (2016-08-13): Fixes #396! I think! Maybe!