phrack/ShootOFF #397: Calibration pattern not going away after auto calibration

cbdmaul, 2016-03-02:
Since I now have to move the arena to the projector for some unknown reason, I have noticed that when I move it, full screen, and it auto calibrates, that the pattern remains.

phrack, 2016-03-02:
Of course not, because we are supposed to never get a release out ever again. I have no idea. It's definitely getting removed with auto calibration and going full screen and starting calibration happens exactly the same way whether the screen is manually or automatically placed.

phrack, 2016-03-02:
I can't dupe this with auto calibration, but I can with manual. If I stop calibration after manual calibration the patter is still there.

phrack, 2016-03-02:
The problem is that exiting full screen doesn't actually stop calibration if it is happening but you want to move the arena. This means then when you full screen again because you're on the right screen you start calibrating again, which saves the pattern as your background like you intentionally set it as the background. Seems like the right answer is to stop calibrating if full screen is exited.

commit 30453693 (2016-03-02): Don't save arena background when calibrating if the pattern is already showing to close #397
closed 2016-03-02 by phrack