phrack/ShootOFF #398: Motion tweaks broke shot detection after calibration

phrack, 2016-03-02:
We really can't catch a break here...

Not sure if it's 

237df50fd2824951b66424fdbc3261599d368628

or 

39695e3af8c34888d4f880752c094596baa29369

Either way, shot detection works fine before calibration but after calibration it doesn't work at all anymore. Even more, everything seems completely shifted off. Note in the attached screenshot how it is complaining about the brightness of the targets but it draws them in completely the wrong place. I duped this in three different sessions and confirmed the camera never moved.

[image]

cbdmaul, 2016-03-02:
that is really strange.  i tested it and it worked fine for me.  neither of those commits changed anything to do with bounds.

cbdmaul, 2016-03-02:
`	// This function ADJUSTS X AND Y FOR LIMITING DETECTION BOUNDS
	private void drawOnFrame(Mat frame, int x, int y, byte[] color) {
		int drawX = x;
		int drawY = y;

		if ((cameraManager.isLimitingDetectionToProjection() || cameraManager.isCroppingFeedToProjection())
				&& cameraManager.getProjectionBounds().isPresent()) {
			drawX = (int) (cameraManager.getProjectionBounds().get().getMinX() + x);
			drawY = (int) (cameraManager.getProjectionBounds().get().getMinY() + y);
		}

		frame.put(drawY, drawX, color);
	}`

This function does not need to adjust for bounds anymore, because processFrame does it.  So we can drop the whole if.

commit c07cdc42 (2016-03-02): Fixes #398 drawing in wrong place
closed 2016-03-02 by cbdmaul