phrack/ShootOFF #400: Stack overflow with shootoff minimized and projector attempting autocalibration

cbdmaul, 2016-03-02:
Exception in thread "JavaFX Application Thread" java.lang.StackOverflowError
	at javafx.beans.property.SimpleBooleanProperty.<init>(Unknown Source)
	at javafx.beans.property.ReadOnlyBooleanWrapper.<init>(Unknown Source)
	at javafx.scene.Node$8.<init>(Unknown Source)
	at javafx.scene.Node.disabledPropertyImpl(Unknown Source)
	at javafx.scene.Node.setDisabled(Unknown Source)
	at javafx.scene.Node.updateDisabled(Unknown Source)
	at javafx.scene.Node.access$500(Unknown Source)
	at javafx.scene.Node$1.invalidated(Unknown Source)
	at javafx.beans.property.ObjectPropertyBase.markInvalid(Unknown Source)
	at javafx.beans.property.ObjectPropertyBase.set(Unknown Source)
	at javafx.scene.Node.setParent(Unknown Source)
	at javafx.scene.Parent$1.onChanged(Unknown Source)
	at com.sun.javafx.collections.TrackableObservableList.lambda$new$19(Unknown Source)
	at com.sun.javafx.collections.ListListenerHelper$Generic.fireValueChangedEvent(Unknown Source)
	at com.sun.javafx.collections.ListListenerHelper.fireValueChangedEvent(Unknown Source)
	at javafx.collections.ObservableListBase.fireChange(Unknown Source)
	at javafx.collections.ListChangeBuilder.commit(Unknown Source)
	at javafx.collections.ListChangeBuilder.endChange(Unknown Source)
	at javafx.collections.ObservableListBase.endChange(Unknown Source)
	at javafx.collections.ModifiableObservableListBase.add(Unknown Source)
	at java.util.AbstractList.add(Unknown Source)
	at com.sun.javafx.collections.VetoableListDecorator.add(Unknown Source)
	at com.shootoff.gui.CalibrationManager.createCalibrationTarget(CalibrationManager.java:127)
	at com.shootoff.gui.CalibrationManager.calibrate(CalibrationManager.java:102)
	at com.shootoff.gui.CalibrationManager.stopCalibration(CalibrationManager.java:71)
	at com.shootoff.gui.CalibrationManager.calibrate(CalibrationManager.java:106)
	at com.shootoff.gui.CalibrationManager.stopCalibration(CalibrationManager.java:71)
	at com.shootoff.gui.CalibrationManager.calibrate(CalibrationManager.java:106)
	at com.shootoff.gui.CalibrationManager.stopCalibration(CalibrationManager.java:71)

cbdmaul, 2016-03-02:
This is actually happening all the time for me.

phrack, 2016-03-02:
It's because removeCalibrationTargetIfPresent doesn't set the optional calibration target to empty. I am fixing it now.

commit 32562bba (2016-03-02): Finish calibration target removal cleanup to close #400
closed 2016-03-02 by phrack