phrack/ShootOFF #402: Color detection is unusually wrong in some scenarios

cbdmaul, 2016-03-03:
Despite being decent in our tests, it is failing on easy cases in the videos provided by /u/greatone123x

I've made a new test from one of his videos.  Will look into it.

[image]
[image]

phrack, 2016-03-05:
I can help test this and maybe try tweaks too. It is wrong about as often as it is right with my projector, I am just rarely using an exercise where it matters. In fact, in most of my uses the color is irrelevant and I don't even look at what they are because I don't show shots on the arena and reset from a target without even looking at my computer.

cbdmaul, 2016-03-05:
Going to push a patch that should fix this pretty well.

[image]

[image]

commit 9da4eada (2016-03-05): Fixes #402
closed 2016-03-05 by cbdmaul