phrack/ShootOFF #407: nu.pattern.opencv is filling people's harddrives

cbdmaul, 2016-03-06:
https://github.com/PatternConsulting/opencv/issues/10

Switch to this to fix:
https://github.com/openpnp/opencv

phrack, 2016-03-06:
Better make sure the fork includes this fix:

https://github.com/PatternConsulting/opencv/issues/11

phrack, 2016-03-14:
Both repositories have the x86 fix so that isn't a concern. However, our gradle config already uses the Maven configuration in the readme. Not sure if the fork that allegedly has this delete file fix is actually in a Maven repository.

phrack, 2016-03-14:
I don't think there is actually a fix out there. The issue linked to at the top guesses that the fork **might** have a fix, but we are already using the version they suggest. We could maybe add a hack just for Windows that looks for old versions of the DLL and deletes them. How are they named on Windows?

cbdmaul, 2016-03-14:
I suggest firing up your vm and looking unless you are okay with waiting a
week.

phrack, 2016-03-15:
I am still looking into this. The file name is the same every time, but the folder name is completely random. We could search the temp folder for the DLL file name, but the file tree gets quite deep and we'd have to search all of it due to the random folder name. This would be a lot of processing for a bug that has nothing to do with us.

My preferred solution is to try to find a hack that allows us to unload the DLL on close, but if that fails the best option may be to just catch the exception, save the full file path, and load the saved path to delete it on the next run.

phrack, 2016-03-15:
I have a decent hack, and there is really no better way to do it (actual bug and solution described below). Basically, on Windows only the best solution at the moment is to redirect the user's default temp directory to a known location for just that instance of ShootOFF, then run a command on shutdown that waits five seconds then deletes that special temporary directory. The reason this is super ugly is that the way to to fork a process that will outlast the JVM for the countdown and delete to actually happen is to cause a separate black command prompt window to pop it up. This ugliness is mitigated by the fact that I can force the window to start minimized (it's never in the way) and it's relatively short lived so it's basically a flash in the task bar. Will clean it up and commit soon.

The problem most basic problem is that the loader that loads the DLL is a static inner class in the OpenCV class. This class doesn't get garbage collected be another static inner classes in the parent class holds a thread object that is used as a shutdown hook to delete the DLL. That thread object that's a shutdown hook can't be garbage collected, thus the parent can't be garbage collected, thus the loader also cannot be garbage collected. The right answer is to split all of those classes up into individual top-level classes.

phrack, 2016-03-15:
Actually, I can hide the prompt entirely, this isn't looking so bad for a hack now!

commit f0f1f1d8 (2016-03-15): On Windows: use a custom temp directory for native binaries and delete stale files on start-up to close #407
closed 2016-03-15 by phrack