phrack/ShootOFF #423: Create Target interface to decouple target classes from GUI

phrack, 2016-03-30:


cbdmaul, 2016-03-30:
As part of the perspective handling class, I was thinking about having an object shim of some sort representing the translation for a specific target.  The target members x,y,height,width wouldn't change, but through the target you could grab this object that tells you what the real size is, what scale it is at, etc etc.  I was then considering some kind of callback that would allow you to quickly rescale all the targets if a perspective parameter changes (i.e. the shooter has changed distances or the camera is moved).

Thoughts? Should the Target object itself implement the callback listener? Keeping in mind that the target members x,y,width,height will change when the target is rescaled.

phrack, 2016-03-30:
Target regions are JavaFX `Node`s and thus already come with all of the transformations you should need: https://docs.oracle.com/javase/8/javafx/api/javafx/scene/Node.html. Thus, instead of a callback I'd use a ReadOnlyProperty, an ObservableValue, or a similar type for a scaling factor stored in the config class. When a TargetView (gui.Target => gui.TargetView when this issue is closed) is created, bind this observable to the scaling property of the target's group. Your perspective class should just update the scaling factor in the config as necessary when the distances change, let the observables plumbing handle the rest. This should effectively confine your changes outside of the perspective class to just a few lines. It will also keep you from creating dependencies between the perspective class and anything but the config.

BTW, I am about to close this issue which is going to drop a bomb on the repo. Lots of small changes, but they go a long way to ensuring that ShootOFF is easier to port while also greatly reducing coupling between components (e.g. almost nothing we'd consider backend is dependent on the GUI anymore).

closed 2016-03-30 by phrack

cbdmaul, 2016-03-30:
We should open a separate issue for this.  I don't think having a single scale setting in the config is going to work, individual targets are going to be at different scales depending on what distance they are set at.  The target should have information akin to "I am set to 10 yards from the shooter" not just "I am scaled at 1.38x to adjust for real world size and .45x to adjust for distance"

phrack, 2016-03-30:
OK, it seems there are multiple requirements here we should tease apart in a separate issue (I'll let you open it since you seem to be claiming it -- which I am OK with):

- Each target must be capable of appearing at a unique distance
- It must be possible for the projector or shooter to change their distance from all targets via user input at any time (I think it's safe to assume that the projector distance -- from camera to projector -- is always the shooter distance for simplicity).
 - This change must be simultaneously reflected in all targets (e.g. all targets get bigger immediately if user is closer)

Based on that it seems like you want each target to know its distance in yards and its scale (it already knows its scale). When a target has its distance set, it is registered with a PerspectiveManager if it has not been registered yet (deregistered if this distance is deleted by the user?). The PerspectiveManager knows the shooters distance is responsible for telling newly registered targets what its scale should be based on both the shooter and the target distance (or the target less the shooter distance?). If the shooter distance is changed it is updated in the PerspectiveManager that knows all targets with set distances and can therefore update them all.

commit a1507bbe (2016-03-30): Decouple target code from GUI to close #423